derdilla/blood-pressure-monitor-fl #8: export time readeable

Psk-Ita, 2023-05-20:
not a bug, but it could be preferibile using ISO formar for CSV export/impkrt, in order to understand the csv file also with a common text editor

NobodyForNothing, 2023-05-20:
From my understanding, you're requesting the time to be displayed in the following format: `YYYY-MM-DDTHH:mm:ss.SSSZ` for improved readability.

I'm uncertain why manual reading is necessary, as the app already provides this information.

My intention behind the export time format was to enable further data analysis and compatibility with other programs, rather than focusing on manual reading.

The primary concern is that offering an option to export in this format would complicate the import process, requiring the app to detect the exported data format.

However, I'm open to further discussion on this matter. Please feel free to provide additional details or expand on your request.

Psk-Ita, 2023-05-20:
Ciao
my needs is to keep a "blood pressure diary" to share with the doctor
I am used to send photos of my handwritten notes

the app is usefull on tracking, let's use this tracks for a real use-case

NobodyForNothing, 2023-05-20:
In the next update the list display range will sync with the graph display range, which can be set to lifetime.

But I will keep the idea of adding different export formats in mind.

closed 2023-05-20 by Psk-Ita