derdilla/blood-pressure-monitor-fl #21: undo after delete

NobodyForNothing, 2023-05-28:
**What you want**
show a snack bar with undo button after entry delete

**Why you want it**
Prevent accidental entry deletion through back gestures in new android version.

uli-on, 2023-05-28:
I just saw, that swiping entries is possible in the new version and wanted to propose a security measurement. I'd prefer a modal dialogue, though: "Do you really want to..."
With an undo bar you have to have your eyes at the screen, while *totally* inadvertently touching the UI would inevitably lead to data loss.

---------------

Two *general* remarks: 
- I love(!)  the way we can enter data, no tabs required, no tapping text fields. 
- And since v1.2.0 even no need to close the virtual keyboard in order to press the Save button.

Thank you!

NobodyForNothing, 2023-05-28:
This is a great idea! I will make it toggle able though as this would get very annoying when you have to delete multiple entries for some reason.

closed 2023-05-29 by NobodyForNothing

uli-on, 2023-05-30:
Thanks, arrived today, much appreciated! :)