derdilla/blood-pressure-monitor-fl #30: Exports for other use

NobodyForNothing, 2023-05-29:
**What you want**
Creating more custom exports that can be read by other software such as ms Excel. This can be related to exports of the data, as shown in the app as a PDF or other widely accepted format.

**Why you want it**
Unfortunately the current time export format isn't supported out of the box and manual time conversions can be prone to errors.

NobodyForNothing, 2023-06-14:
Also plan to add formating options for export

NobodyForNothing, 2023-06-23:
pdf still not implemented

uli-on, 2023-07-26:
A human readable date is also something that doctors and their employees would profit from.

I'd like to propose a divider here on the sorting panel. Items above the divider are exported, items below are left out. Maybe someone don't want their comments be read by the surgery team.

And thanks for your great efforts around exporting!

NobodyForNothing, 2023-07-26:
You can swipe the elements in the sorting panel to the side to remove them from export. This also allows you to export time in ISO format, which is easier to read.

As you are now the second person who didn't notice this, I will consider using your design.

uli-on, 2023-07-26:
If I interpret you right, this would mean with the ISO date visible right from the start, below the divider, and selectable by dragging upwards. That'd be cool!

NobodyForNothing, 2023-07-30:
If anyone wants something to be part of the PDF export, here is the time and place to make your suggestions. Let me emphasize, that the export-import system should be as comprehensive as possible.

It is currently planned to have the following components available:
- [ ] A list of all measurements
  -  [ ] Custom field order like CSV
- [ ] A start sheet with important information
  -  [ ] average/max blood pressure
  -  [ ] blood pressure at different times of the day (morning, evening, ...)
  -  [ ] user addable labels where they can put information like name, doctor, ...
  -  [X] start and end date of export data

The current state of the PDF export: (will update)
[image]

uli-on, 2023-07-30:
The table page would offer room for one more column: Pulse pressure. This is calculated from the difference between systole and diastole and is the measure for the tightness of the heart valves.

uli-on, 2023-07-30:
My main concern would be the export to CSV, though: Is it perhaps possible to fine-tune the date a bit further, so that post-processing and regex with a text editor on a desktop PC plus transfers to and fro become unnecessary? For making the file compatible with my analysis app, I myself would have to replace the „T“ of the ISO time with a blank space, remove the decimal places from the time (dot as well as thousandths) and append a few more empty fields (more precisely `null,,0.0,0`). Cause the empty fields in my case would be appended in one piece (and not injected here and there), I needed one extra custom field that I could define myself.

NobodyForNothing, 2023-07-30:
@uli-on What analysis app are you using? I would probably add an export format for it so others have a preset they can easily use and I can avoid adding a bunch of generic options that will only be useful for one app. Would you like this solution or do you want something more generic?

I will allow adding custom columns for things like pulse pressure, and maybe allow adding your own fields with basic math.

uli-on, 2023-07-30:
I'm using My Heart/Blutdruck (unfortunately from the Google Play Store), currently parallel to Blood Pressure Monitor. It has a plethora of computations it conducts on these three basic values and creates n kinds of diagrams from it (but is quite sluggish starting up and loading data and uses these awkward huuuge wheels to enter values).

A sample of its data:

```
DATUM,SYSTOLE,DIASTOLE,PULS,Beschreibung,Tags,Gewicht,Sauerstoffsättigung
2023-07-30 13:21:45,118,76,60,null,,0.0,0
```

I could totally stop feeding it with data if I could –  once a month or like so – use the export from *your* app, which is so much quicker and easier to fill in. 

> I would probably add an export format for it so others have a preset they can easily use and I can avoid adding a bunch of generic options that will only be useful for one app. Would you like this solution or do you want something more generic?

Yes, for my own interest this would be ideal, thanks!! This would not be cast in stone, right, so that others could also exchange/rearrange elements for *their* export interests?

NobodyForNothing, 2023-07-31:
> > I would probably add an export format for it so others have a preset they can easily use and I can avoid adding a bunch of generic options that will only be useful for one app. Would you like this solution or do you want something more generic?

> Yes, for my own interest this would be ideal, thanks!! This would not be cast in stone, right, so that others could also exchange/rearrange elements for their export interests?

Yes exactly. I would add the option to add your own fields to the CSV file (For example Title: *Gewicht* Value: *0.0* or Title: *DATUM* Value: *$DATEFORMAT(yyyy-MM-dd HH:mm:ss)*). These options will be documented and I try to keep them minimal.

As these take effort to set up,  I will also include a list of presets (e.g. Default, My Heart Export, ...). I would also write something in there to encourage sharing a configuration here and have it added to the defaults.

uli-on, 2023-08-03:
If this is an excerpt from the exported csv, then you're already there in the first go. Great!
No editor tricks, no. I would have to use the editor only in case you hadn't succeeded with your csv export. But that's not required, now, no duplicate apps any longer, no file transfers and no regexing. Thank you!! 👍

closed 2023-08-09 by NobodyForNothing