derdilla/blood-pressure-monitor-fl #37: Prepared for translation?

uli-on, 2023-05-30:
Is your app set up to be translated? I've accounts for Crowdin, Weblate and Transifex and could supply the German version.

NobodyForNothing, 2023-05-30:
The project currently is not set up for translation and flutter doesn't support it out of the box either (https://docs.flutter.dev/accessibility-and-localization/internationalization).

I am not sure if this is really necessary. In my perception many projects have translations although they don't really need them.

At some point I plan to add it, even if its only for keeping the strings in one place (I can translate it to german as well)

If you want to setup the infrastructure now, sure go ahead 😉

uli-on, 2023-05-30:
I'm not keen on having German-only apps, I just thought of elder people and those less used of reading English. 
Re infrastructure: Like I wrote later on in one of the closed topics: I'm not a dev. I just wanted to contribute what I can. And that way compensate for the hassle I'm causing with bug reports (and the feature ideas I'll likely post).

NobodyForNothing, 2023-05-30:
Don't worry, bug reports and feature requests are always welcome its better to have them now than in a few years when I would have to get into the code again just to fix them.

uli-on, 2023-05-30:
Ah, just opened this page from the pc and saw your location. I usually visit Github via Octodroid, no need to log in and out and can't be MS-tracked either, but also no location entries. Rather a forum app for bug reports and ideas.

uli-on, 2023-05-31:
Oops, didn't see you labeled the topic today. I leave it to you then.

NobodyForNothing, 2023-06-09:
Localization infrastructure has been put in branch. Now its just a matter of copying the text from the app, adding references to the text and translating.

uli-on, 2023-06-10:
Just wanted to signalise that I didn't withdraw my offer, so if you like, you can let me do the translation and only revise the outcome.

NobodyForNothing, 2023-06-10:
If you want to help, you can try to translate the values from `lib/l10n/app_en.arb` to `lib/l10n/app_de.arb` in the `feat-localization` branch.

The most important thing is to bring the message across, but you can also try to:
- sound friendly / happy (error messages shouldn't discourage)
- be brief (when the translation is longer than the english version, there can be some field size errors that need to be manually resolved)

uli-on, 2023-06-10:
You've randomly left out lines in the DE file before stopping somewhere after 1/4 with "SAVE". That makes it a speculation for me, what's intentional, what's to leave out, ... . I can also download the EN file and translate/save as app_en.arb, but I'd prefer you complete the DE file with the now missing English strings, and then I gladly do the rest. Is that OK for you?

NobodyForNothing, 2023-06-10:
done. You don't need to do anything for the templates (@)

uli-on, 2023-06-10:
It wouldn't make sense for me to learn how to do a pull request. I simply paste the arb-file here.
I added some comments, but these are clearly visible.

```
{
  "title": "Blutdruck app",

  "success": "Erfolg: {msg}",
  "@success": {
    "placeholders": {
      "msg": {
        "type": "String"
      }
    }
  },
  "loading": "Lade...",
  "error": "Fehler: {msg}",
  "@error": {
    "placeholders": {
      "msg": {
        "type": "String"
      }
    }
  },
  "errNaN": "Bitte Zahl eingeben",
  "errLt30": "Wert < 30? Deaktiviere Prüfung in Einstellungen!",
  "errUnrealistic": "Fehleingabe? Deaktiviere Validierung in den Einstellungen!",
  "errUnknown": "Unbekannter Fehler",
  "errCantOpenURL": "Kann URL nicht öffnen: {url}",
  "@errCantOpenURL": {
    "placeholders": {
      "url": {
        "type": "String"
      }
    }
  },
  "errNoFileOpened": "Keine Datei geöffnet",
  "errNotStarted": "Nicht begonnen'",
  "errNoValue": "Bitte Wert eingeben",
  "errNotEnoughDataToGraph": "Zuwenig Daten für Graphen",
  "errNoData": "Keine Daten",

  "btnCancel": "ABBRUCH",
  "btnSave": "OK",
  "btnConfirm": "OK",
  "btnUndo": "ZURÜCK",

  "sysLong": "Systolisch",
  "sysShort": "sys",
  "diaLong": "Diastolisch",
  "diaShort": "dia",
  "pulLong": "Puls",
  "pulShort": "pul",
  "addNote": "Notiz (optional)",

  "layout": "Layout",
  "allowManualTimeInput": "Editierbare Zeitangaben",
  "enterTimeFormatScreen": "Datums-/Zeitformat",
  "followSystemDarkMode": "Thema wie System",
  "darkMode": "Dunkles Thema",
  "iconSize": "Größe der Buttons",
  "graphLineThickness": "Linienstärke d. Graphen",
  "animationSpeed": "Animationsdauer",
  
  
  #################################################################
  Neuer String, bitte selbst übersetzen
  #################################################################
  "graphTitlesCount": "graph label count",
  "accentColor": "Hauptfarbe der App",
  "sysColor": "Farbe für Systole",
  "diaColor": "Farbe für Diastole",
  "pulColor": "Farbe für Puls",
  "behavior": "Verhalten",
  "validateInputs": "Eingaben prüfen",
  "confirmDeletion": "Löschen bestätigen",
  "age": "Alter",
  "ageDesc": "Bestimmt die Warnwerte",
  "aboutWarnValuesScreen": "Hilfe",
  "aboutWarnValuesScreenDesc": "Mehr Infos zu den Warnwerten",
  "overrideWarnValues": "Warnwerte außer Kraft setzen",
  "sysWarn": "Warnwert Sys",
  "diaWarn": "Warnwert Dia",
  "data": "Daten",
  "useExportCompatability": "Kompatibler Export",
  "useExportCompatabilityDesc": "Exportiert MIME-type als Text",
  "export": "Exportieren",
  "exportSuccess": "Exportiert in: {path}",
  "@exportSuccess": {
    "placeholders": {
      "path": {
        "type": "String"
      }
    }
  },
  "shared": "Geteilt",
  "import": "Import",
  "sourceCode": "Quellcode",
  "licenses": "Drittpartei-Lizenzen",

  "statistics": "Statistik",
  "measurementCount": "Zahl der Messungen",
  "measurementsPerDay": "Messungen pro Tag",
  "timeResolvedMetrics": "Werte nach Uhrzeit",
  
  
  ######################################################
  Textfelder sind sehr eng, daher das kürzere math. Symbol. 
  UNICODE U+2300, DEZIMAL &#8960;
  ######################################################
  "avgOf": "{txt} ⌀",
  "@avgOf": {
    "placeholders": {
      "txt": {
        "type": "String"
      }
    }
  },
  "minOf": "{txt} min.",
  "@minOf": {
    "placeholders": {
      "txt": {
        "type": "String"
      }
    }
  },
  
  
  ###########################################
  Hier stand ursprüngl. "may."
  ###########################################
  "maxOf": "{txt} max.",
  "@maxOf": {
    "placeholders": {
      "txt": {
        "type": "String"
      }
    }
  },

  "warnValues": "Warnwerte",
  "warnAboutTxt1": "Die Warnwerte sind ledigl. Vorschläge, kein medizinischer Rat.",
  "warnAboutTxt2": "Die voreingestellten Basiswerte für Altersstufen stammen aus dieser Quelle.",
  "warnAboutTxt3": "Ändere die Werte nach Bedarf, aber befolge die Empfehlungen deines Arztes.",

  "enterTimeFormatTxt1": "Ein Formatier-String besteht aus einer Mischung vordefinierter ICU-Strings sowie anderer Zeichen deiner Wahl.",
  "enterTimeFormatTxt2": "Die vollständige Liste gültiger Formate findest du hier.",
  "enterTimeFormatTxt3": "Bitte beachte, dass kürzere/längere Formatier-Strings nicht die Spaltenbreiten ändern. Sie führen eher zu unerwünschten Umbrüchen.",
  
  
  ################################################
  M. E. ist das Zeitformat jetzt HH:mm (zwei H, betr. auch EN)
  ################################################
  "enterTimeFormatTxt4": "Voreingestellt: \"yy-MM-dd H:mm\"",
  "enterTimeFormatString": "Zeitformat-String",

  "dateFormatting": "Datumsformat",
  "now": "jetzt",
  "notes": "Notizen",
  "time": "Zeit",
  "confirmDelete": "Löschen bestätigen",
  "confirmDeleteDesc": "Soll der Eintrag wirklich gelöscht werden? (Diese Rückfrage kann in den Einstellungen abgestellt werden.)",
  "deletionConfirmed": "Eintrag wurde gelöscht"
}
```

uli-on, 2023-06-10:
BTW, I didn't correct the app's name as this is the brand. I leave it to you.

NobodyForNothing, 2023-06-11:
Good catch with the max and the changed date format. Most of translations are accurate and fit in the boxes. I added the big avg symbol (Ø) to the English version as well. You can see the result on the branch. Feel free to criticize 😉.

Since you didn't add a PR, you wont be listed as a contributor automatically. What kind of attribution do you want?

uli-on, 2023-06-11:
> Good catch with the max

Confusing that it didn't show up inside the app, it was also in EN `*shrug*`

> Feel free to criticize 😉.

Hope you don't perceive my proposal(s) as criticism.
While we're on the subject, the new string `anzahl graph Beschriftungen` can be adjusted yet :) What I'd be interested to hear: Did I leave the file syntax intact (except for the comments)? 

> What kind of attribution do you want?

None, thanks. I'll have the result, that's enough appreciation. BTW, will the language be dependent on the language choice in Android or will there be an app internal switch?

NobodyForNothing, 2023-06-12:
> Did I leave the file syntax intact (except for the comments)?

Your file syntax worked. There were some empty lines around the comments but that's all.

> Will the language be dependent on the language choice in Android or will there be an app internal switch?

In the current implementation it is dependent on the android language, as someone who needs it might not be able to change it in the settings. I currently don't see a reason to add a in app option for this.

uli-on, 2023-06-12:
I was mainly interested in how the language switch is going to happen technically and hoped Flutter would have a drop-in "snippet" for that scenario. It's easier for prospective translators to see the original strings in place without getting English search engine results and other lang specific web & app contents, but it's not necessary.

uli-on, 2023-07-26:
@NobodyForNothing
Bist du einer der Dillas auf Weblate (welcher) oder ein "nobody" ? Hätte dort eine Anmerkung zum Englischen mitzuteilen, und kurze Wege sind sowieso angenehmer.

NobodyForNothing, 2023-07-26:
@uli-on Ich bin dort `derdilla06`. Meistens wird das suggestion feature genutzt, aber schieß los.