derdilla/blood-pressure-monitor-fl #46: Specify dates presented on stats page

NobodyForNothing, 2023-06-06:
**What you want**
Ability to specify the interval in which the data is collected an the statistics page.
Show metrics resolved on day a week to see differences on weekends.

**Why you want it**
Useful for viewing developments in data, problematic times.

tb06, 2023-07-15:
Hello:

I'm not sure I understand perfectly the feature you request but I think that it's what I need too:

In France (and apparently in other countries), the auto-measurement process requires to take several measures during 3 consecutive days (3 measures each morning, 3 measures each evening).
What would be really useful in the app would be the ability to get the average blood pressure information for these 3 days only (and maybe a special report to view and export only the measures of these 3 days and the related average figures).

If it's another request than this one, I can create another issue.
Thanks in advance!

NobodyForNothing, 2023-07-15:
@tb06 this results in 3 features actually ;) but don't worry I can handle it.

The way I understand this,
1. you want to be able to fully specify the range in which data is queried (created #97)
2. you want to have the data-range selection on the stats page as well as on the export page (this issue)
3. you want PDF export (#30)

If I got anything wrong, feel free to correct me.

1 and 2 can be done relatively quickly and will be in the next release, while 3 might take some more time.

tb06, 2023-07-15:
> @tb06 this results in 3 features actually ;) but don't worry I can handle it.
> 
> The way I understand this,
> 
>     1. you want to be able to fully specify the range in which data is queried (created [manual data range option #97](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/issues/97))
> 
>     2. you want to have the data-range selection on the stats page as well as on the export page (this issue)
> 
>     3. you want PDF export ([Exports for other use #30](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/issues/30))
> 
> 
> If I got anything wrong, feel free to correct me.
> 
> 1 and 2 can be done relatively quickly and will be in the next release, while 3 might take some more time.

Hello @NobodyForNothing:

You're perfectly right, thanks a lot!

Thomas.

closed 2023-07-18 by NobodyForNothing