derdilla/blood-pressure-monitor-fl #50: Auto export

khao-soi, 2023-06-10:
**What you want**
Auto export function to a pre-defined location. Triggered after adding a measurement. 

**Why you want it**
To prevent data loss. If something happens to your device and you haven't exported your data to an external location for a while, its all gone. 
With auto export you could just configure your measurements beeing exported automatically. See OpenTracks as a good example for this.

NobodyForNothing, 2023-06-11:
I can see why you want to avoid data loss, but I don't see how the [OpenTracks](https://github.com/OpenTracksApp/OpenTracks) solution really helps. 

Exporting the data to a location on your device will in most cases not make it anymore accessible. The only scenario I can think of where the proposed solution would help is the display getting damaged after setting the export location on a SD-card. In the majority of situations (e.g phone gets stolen / falls into a river) a local export would not really improve the situation. Whats even worse is if there is a regular export set up, someone stealing the phone now has access to what is essentially your latest medical information.

To prevent data theft I would need to let the user encrypt the data, which would make using exports harder and more complicated for the average user and might lead to data loss anyways, if the user forgets their password. The storage location is an even harder problem to solve. I can only think of 2 other places where the data can be stored:

- a central cloud storage like firebase or dropbox - this is probably the most accessible solution but it has the problem of depending on non-free services to which I have to maintain a relation and you have to trust your data to, as well as the legal rabbit hole that comes with exporting data out of the EU / other local legislation
- a self hosted server - this is probably the most data friendly solution but it would not be accessible for most users and would probably require you to set up ssl certificates. This will lead to a scenario where nobody uses it and bugs wont get caught in time

In the end these are just blood pressure values so I don't think high availability is really justified. Additionally this amount of bloat will inevitably cause bugs, security issues and data loss and drive up development effort drastically

If you know any other solutions or reasons why you want auto-export regardless feel free to reopen the issue.

khao-soi, 2023-06-11:
I should have mentioned that I auto export the data to folders that are synced with my nextcloud.

NobodyForNothing, 2023-06-11:
While this isn't a safe practice, I will reconsider it. I have to check what kind of permissions this will require.

closed 2023-06-26 by NobodyForNothing