derdilla/blood-pressure-monitor-fl #57: Wrong number of days in week view

thatso, 2023-06-17:
**What happened**
While the headline in week view shows 8 days, the table lists only 6 days:

[image]

**ideas**
A week view should show exactly 7 days.
Notice in the screenshot that the dates 23-05-22 to 23-05-29 span 8 days while the table stops at 23-05-27 instead of 23-05-28.
Also, it is not clear if the headline is meant as legend for the diagram above. Does the additional day come from the fact that the week should stop at 23-05-28 23:59 instead of 23-05-29 00:00? Maybe leave out the time altogether as it adds no informational value?

NobodyForNothing, 2023-06-17:
Hello,
thank you very much for reporting this bug. You are very right that `23-05-29 00:00` is more confusing than `23-05-28 23:59`. I will change it to the latter in the next update (at least visually to avoid having blind spots)

The 2 timestamps signal the range in which data is queried, they are there to avoid confusion on the slightly ambiguous words for the range definition (week could be interpreted as *last 7 days*) and to indicate which week is shown. The information the time provides is mainly relevant for the day view. 

I can't confirm the issue with showing just data from 6 days. Are you sure you have stored an entry on the `23-05-28` could you check for it in a wider range again and make sure it's not just scrolled away? If you can confirm there is an entry in that time range that is not shown, please provide the version of the app you are using (it's in the android app-info).

NobodyForNothing, 2023-06-17:
When sharing screenshots of the app I generally recommend overriding the pixels containing measurements as it has been shown that pixelated text can be deciphered. Especially when its just a small set of numbers.

thatso, 2023-06-17:
The missing day in the week view is present for all weeks. Surely, I've checked that data is present for every single day. There is nothing to scroll. Actually, there is a lot of usable screen space left on my display:

[image]

While there is indeed no data for 2023-05-20, you can see in this screenshot that the two entries for 2023-05-21 are displayed, resulting in 7 table rows.

Maybe you could simply state the calendar week number to make the time range clear in the headline as the day dates are in the list anyway? I am aware that for different countries the week starts on Sunday or Monday. I expect the correct value can be retrieved from Android system settings? 

My version is 1.3.0 from Fdroid.
BTW: the "about" item in the settings shows information about the warn values. While this is explained in the description below the menu entry, one would expect to see the program version which currently can only be retrieved in the Android app info.

NobodyForNothing, 2023-06-18:
If the problem is that there is no data displayed for the last day of the interval (in your second image it would be 2023-05-22), that is a communication problem that will be fixed in a upcoming update through either removing a second from end date or a interval specific solution like the displaying week as  you suggested.

I still stumble across the second part of your bug report:
> the table lists only 6 days

If you have a problem with entries that are actually in the interval but not displayed could you maybe provide dates where this is the case.

thatso, 2023-06-19:
Strange ... I've rebooted my phone and can no longer reproduce the missing day in week view. :astonished:
Closing this issue for now, sorry for any inconvenience.

closed 2023-06-23 by NobodyForNothing