derdilla/blood-pressure-monitor-fl #74: import doesn't work

kyrawertho, 2023-06-26:
**What happened**
I want to use my existing data but I can't import it, so I tried to import an exported file, which also doesn't work:

- Add some data
- Export to file
- Clear all data
- Import exported file
-> Nothing is imported.

**ideas**

NobodyForNothing, 2023-06-26:
Thanks for the bug report. Importing data from older versions is currently broken and will be fixed in the next release. 

I can't confirm the more general issue you describe. Could you please provide you export configuration and say what the bar at the bottom sais after attempting import.

kyrawertho, 2023-06-26:
I don't see a bottom bar. Nothing at all happens when i try to import.

timestampUnixMs, systolic, diastolic, pulse, notes
1679860800000,"107","62","","example"
1679914800000,"192","84","",""

NobodyForNothing, 2023-06-27:
Did you manually change the exported file? The current version does not support this. But it will be fixed once 1.3.3 is released. (might take a bit)

kyrawertho, 2023-06-27:
No, importing the exported file unchanged doesn't work either. I don't have much use for the app if I can't import tbh.

Android 8.1.0

NobodyForNothing, 2023-06-27:
What version of the app are you using? (can be found in android app info; the next version will be more clear about that) And could you provide the configurations you used for export (assuming you already use the latest 1.3.2)?

BackSeat, 2023-07-03:
I'm trying to import a file that starts like this:

```
timestampUnixMs,systolic,diastolic,pulse,notes
1688372700000,122,81,76,
1687769640000,123,81,73,
1687165260000,114,80,76,
```

I get, "This file can't be imported"

v1.3.2 from f-droid.

Is there any more info that would be helpful?

NobodyForNothing, 2023-07-12:
@BackSeat In version v1.3.3 the data you provided is importable. Please test again once the F-Droid build is ready.

NobodyForNothing, 2023-07-12:
@kyrawertho The data you provided still doesn't work because there is no pulse data present (support for incomplete data is being discussed in #75). 
Could you please check if importing an unchanged file works now in v1.3.3 (you might need to wait for F-Droid)?

NobodyForNothing, 2023-07-12:
> But now you just have to guess where to tap when using dark mode: 

This should be fixed as well. Are you sure that you are using v1.3.3? 
If so: do you use any system-wide dark mode features that might force this?

kyrawertho, 2023-07-12:
Sorry, I should have checked. I got the latest one off Fdroid, which is 1.3.2. My OS doesn't have dark mode features.