derdilla/blood-pressure-monitor-fl #80: Support Bluetooth LE blood pressure monitor devices

andreassch, 2023-07-04:
It would be great to support Bluetooth LE blood pressure monitors via the standard GATT protocol. This would reduce the chances for typos while entering the values as well as enhance comfort. Moreover, such a feature might get users away from commercial apps shipped with devices.
If I understand correctly, the app needs to subscribe to the Blood Pressure GATT service (UUID 0x1810) with the Blood Pressure Measurement characteristic (UUID 0x2A35) and potentially the Date Time characteristic (UUID 0x2A08). The Blood Pressure Measurement characteristic is described in Section 3.31 of the [GATT Specification Supplement](https://www.bluetooth.org/docman/handlers/DownloadDoc.ashx?doc_id=555004).

NobodyForNothing, 2023-07-05:
I agree that there are few non-commercial 
apps for BLE-devices and can see why such a app would be useful. 

But I am not quite sure if it would fit into this app. The main goal is to have long time monitoring of blood pressure, pulse data is commonly recorded during activities.

How exactly would you want to use this feature?

andreassch, 2023-07-05:
I would want this as an alternative measurement entry method, for example, with an additional +-Bluetooth button (showing a + with the Bluetooth symbol on the button) above or below the current + button. Alternatively, the data entry method for the (single) + button (manual or Bluetooth) might be selected in the settings. (The Bluetooth device provides the same data that are now entered manually, i.e. systolic and diastolic blood pressure and pulse.)
Additionally, an option to pair/unpair a device would be need e.g. in the settings.

NobodyForNothing, 2023-07-05:
Ok, thanks for clearing that up. 
Although the feature seems useful and I plan to add it you might have to wait a bit, as there are some other more urgent things I need to take care of first and won't be able to test on a real device for now.

andreassch, 2023-07-05:
Thinking more about it, I realize a new button is not needed. When a device is paired and has subscribed to the blood pressure characteristic, it is notified about a new measurement and can take it over, potentially showing a dialog with the values and an option to enter a note.

andreassch, 2023-07-05:
I can offer to test with a device.

zanovis, 2024-02-20:
> I can offer to test with a device.

@andreassch I agree that supporting BLE data import would be a great feature. Which device model do you use/recommend? I may be able to help with this.

NobodyForNothing, 2024-03-21:
Sorry for taking so long on this, but now I have some capacity to get into this. For gathering information I compiled the sample app from [reactive ble](https://github.com/PhilipsHue/flutter_reactive_ble/)  and would like if someone who has a BLE GATT device could check what data they can get through verbose logging. 

About using the app:  Unzip the zip archive ([reactive_ble_TEST.zip](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/files/14696846/tst.zip)) and install the apk. You need to first allow location in the android app settings. To perform a scan you need to press the button.
When the UUID field is empty all available devices should be shown, but it would be important to hear if inputting `1810` (or `180D`) in the service UUID field also shows the device. These values represent the Blood Pressure (Hear Rate) Service according to the [GATT UUID specification](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Assigned_Numbers/out/en/Assigned_Numbers.pdf?v=1711031296353).

On the connect to device screen please report all available information and the logs so I have a base for my development. (black out all potentially collected measurements)

andreassch, 2024-03-21:
I have compiled the reactive ble test app and installed it on my phone. When I type `1810` in the UUID field, my device is shown while scanning. I can connect. Which logs are you referring to, and where can I find them?
(I previously used the [nRF Connect app by Nordic Semiconductor](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) to check the capabilities of my device.)

NobodyForNothing, 2024-03-22:
@andreassch Thanks so far, activate verbose logging, then tap on your device. The following screen has 2 tabs one with a connect button and another one with the logs.

andreassch, 2024-03-24:
Thanks for the hint. Unfortunately, there is not much in this tab (which is why I first thought there would be an additional log):
```
19:51:11.847 - Start ble discovery
19:51:52.543 - Stop ble discovery
19:51:54.069 - Start connecting to xx:xx:xx:xx:xx:xx
19:51:54.102 - ConnectionState for device xx:xx:xx:xx:xx:xx : DeviceConnectionState.connecting
19:51:55.531 - ConnectionState for device xx:xx:xx:xx:xx:xx : DeviceConncectionState connected
```
On the other hand, nRF Connect gives the following:
```
nRF Connect, 2024-03-22
boso medicus CE6674  (xx:xx:xx:xx:xx:xx)
I	10:49:38.952	[Server] Server started
V	10:49:38.985	Heart Rate (0x180D)
- Heart Rate Measurement [N] (0x2A37)
   Client Characteristic Configuration (0x2902)
- Body Sensor Location [R] (0x2A38)
- Heart Rate Control Point [W] (0x2A39)
Unknown Service (0000aaa0-0000-1000-8000-aabbccddeeff)
- Unknown Characteristic [N R] (0000aaa1-0000-1000-8000-aabbccddeeff)
   Client Characteristic Configuration (0x2902)
   Unknown Descriptor (0000aab0-0000-1000-8000-aabbccddeeff)
   Characteristic User Description (0x2901)
   Characteristic Presentation Format (0x2904)
- Unknown Characteristic [I W WNR] (0000aaa2-0000-1000-8000-aabbccddeeff)
   Client Characteristic Configuration (0x2902)
User Data (0x181C)
- First Name [R W] (0x2A8A)
- Last Name [R W] (0x2A90)
- Gender [R W] (0x2A8C)
Environmental Sensing (0x181A)
- Carbon Monoxide Concentration [N R] (0x2BD0)
   Client Characteristic Configuration (0x2902)
V	10:49:39.433	Connecting to xx:xx:xx:xx:xx:xx...
D	10:49:39.433	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	10:49:39.513	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	10:49:39.513	[Server] Device with address xx:xx:xx:xx:xx:xx connected
D	10:49:39.525	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	10:49:39.525	Connected to xx:xx:xx:xx:xx:xx
D	10:49:39.526	wait(1600ms)
D	10:49:39.558	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	10:49:39.915	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
D	10:49:39.935	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDED (12)
I	10:49:39.935	Device bonded
V	10:49:41.126	Discovering services...
D	10:49:41.126	gatt.discoverServices()
D	10:49:41.196	[Callback] Services discovered with status: 0
I	10:49:41.196	Services discovered
V	10:49:41.242	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Privacy Flag [R W] (0x2A02)
- Reconnection Address [W] (0x2A03)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Blood Pressure (0x1810)
- Blood Pressure Measurement [I] (0x2A35)
   Client Characteristic Configuration (0x2902)
- Blood Pressure Feature [R] (0x2A49)
- Date Time [R W] (0x2A08)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- Serial Number String [R] (0x2A25)
- Hardware Revision String [R] (0x2A27)
- Firmware Revision String [R] (0x2A26)
- Software Revision String [R] (0x2A28)
- System ID [R] (0x2A23)
- IEEE 11073-20601 Regulatory Certification Data List [R] (0x2A2A)
Battery Service (0x180F)
- Battery Level [R] (0x2A19)
Unknown Service (233bf000-5a34-1b6d-975c-000d5690abe4)
- Unknown Characteristic [R W] (233bf001-5a34-1b6d-975c-000d5690abe4)
D	10:49:41.243	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	10:49:41.246	gatt.setCharacteristicNotification(00002a35-0000-1000-8000-00805f9b34fb, true)
I	10:49:49.810	Connection parameters updated (interval: 198.75ms, latency: 0, timeout: 6000ms)
D	10:49:50.779	[Callback] Connection state changed with status: 19 and new state: DISCONNECTED (0)
W	10:49:50.779	Connection terminated by peer (status 19)
I	10:49:50.779	Disconnected
D	10:49:50.780	[Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	10:49:50.780	[Server] Device disconnected
D	10:49:51.055	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```
Hope that helps.

NobodyForNothing, 2024-03-31:
@andreassch I build a debug version that _should_ already display measurements taken through Bluetooth (the button is at the top of the add measurement dialogue).

### Attention
This is a debug build. When installing it you *will* use all data stored in the app. It is signed using the debug keys so it might only be installable through [adb](https://source.android.com/docs/setup/build/adb#download-adb). If that is a problem for you, I can provide a version signed with the normal key (could be confusing in case the apk spreads).

[READ THE DISCLAIMER (download)](https://drive.google.com/file/d/1terx8B_f0j1r2Egn9d6EI9LbfhgMTnJt/view?usp=sharing)

### Feedback
It would be great if you could confirm whether the measurement gets correctly transmitted and decoded. Please include your Android version and device/android variant. In case you encounter any bugs (or red and yellow errors) please report them as well.

In the final version, the form would be autofilled with the received values and show applicable warnings (wrong measurement position etc.) as an information bar.  In case you have any thoughts on how UX / UI could be improved feel free to share them right now.

andreassch, 2024-04-09:
Thank you very much for implementing this feature. I have compiled the app from the branch 80-support-bluetooth-le-blood-pressure-monitor-devices and installed it on my phone. When pressing the Bluetooth button in the add measurement dialogue, it scans and finds my device when it is active (I take that from the scanning indication in the Bluetooth field vanishing, leaving the field empty). However, no values are taken over, nothing happens. What is the best way to debug?

NobodyForNothing, 2024-04-10:
If I understand you correctly you pressed the Bluetooth icon once and then enabled the measurement device before the field went blank? At that stage there _should_ be a list of found devices, there is no step where the field should "vanish". I just pushed another commit that ensures the list has enough height.
 
> What is the best way to debug?

If you feel comfortable with [logcat](https://developer.android.com/tools/logcat) you can provide logs. When no errors are displayed, you could go further and add `print()` statements and tweak the code in `lib/components/ble_input/` (relatively easy in Dart, even without any experience).

Don't feel pressured: Explaining what screens were shown and which steps you took is already quite helpful.

andreassch, 2024-04-10:
With your new code, the behaviour is still the same. I have attached two screenshots to demonstrate it. The first one shows the app when scanning without the device available:
[image]
The second one shows it when the device has become available:
[image]
The same shows when starting scanning while the device is available (it probably recognizes it so fast that I don't see the scanning screen).
I also tried logcat via
```
adb logcat flutter:V BluetoothLeScanner:V
```
but it didn't show anything that seemed useful to me.
Can you please name code parts where you would add debug statements or add them there yourself?

NobodyForNothing, 2024-04-11:
>  the behaviour is still the same

That's really odd, are you sure you pulled the latest changes before building? I can't find any data that would trigger this behavior of the UI. I added a 'copy debug' button below the Bluetooth input that copies any relevant errors to the clipboard.

> Can you please name code parts where you would add debug statements or add them there yourself?

I added some to the `ble_input.dart` file. As this file contains the UI code of the component, data causing render errors would be passed through there. The logic of measurement taking is handled by `ble_input_bloc.dart`.

andreassch, 2024-04-12:
Thank you for adding the debug code. Whether I made a mistake while installing the new version last time, I cannot tell. Nevertheless, with your new code I now see the 'copy debug' button. When activating my device, I see it three times:
[image]
Using the 'copy debug' button, I get the following log:
```
STATE:BleInputClosed
STATE:BleInputLoadInProgress
STATE:BleInputLoadInProgress
STATE:BleInputLoadSuccess
BleInputLoadSuccess:[DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -55, connectable: Connectable.available)]
STATE:BleInputLoadSuccess
BleInputLoadSuccess:[DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -55, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -58, connectable: Connectable.available)]
STATE:BleInputLoadSuccess
BleInputLoadSuccess:[DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -55, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -58, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -67, connectable: Connectable.available)]
STATE:BleInputLoadSuccess
BleInputLoadSuccess:[DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -55, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -58, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -67, connectable: Connectable.available)]
[...]
STATE:BleInputLoadSuccess
BleInputLoadSuccess:[DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -55, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -58, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -67, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -57, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -66, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -54, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -56, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -63, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -64, connectable: Connectable.available)]
STATE:BleConnectInProgress
STATE:BleInputLoadSuccess
BleInputLoadSuccess:[DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -55, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -58, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -67, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -57, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -66, connectable: Connectable.avail [... cut ...] edDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -53, connectable: Connectable.available), DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -59, connectable: Connectable.available)]
STATE:BleConnectFailed
```
At the end, the device switched itself off.
Does that log help?

NobodyForNothing, 2024-04-13:
(Yes, that log did help). I fixed the duplicated entries and some other bugs that could cause the phases to change unexpectedly. So if I understood you correctly you opened the input and selected one of the three devices (should all have worked) and then your device switched off? It would be great if you could be more precise in this part (e.g. I did X then saw Y).

On an unrelated note, I was wondering whether selecting from a List of devices is even useful. This could be implemented in a way that the measurement with the first found device. I would appreciate your input on this.

andreassch, 2024-04-13:
Sorry that my last report was not precise enough. I clicked on an entry, and when nothing happened after about half a minute or a minute, I clicked on the next entry, and then after more waiting on the third entry. Finally, some time after clicking on the third entry, the device switched itself off due to the normal timeout.

With the new version, the scanning never finds my device. I get the following log from 'copy debug':
```
STATE:BleInputClosed
STATE:BleInputLoadInProgress
PLATFORM: {{'package:bloc/src/emitter.dart': Failed assertion: line 114 pos 7: '!_isCompleted': 

emit was called after an event handler completed normally.
This is usually due to an unawaited future in an event handler.
Please make sure to await all asynchronous operations with event handlers
and use emit.isDone after asynchronous operations before calling emit() to
ensure the event handler has not completed.

  **BAD**
  on<Event>((event, emit) {
    future.whenComplete(() => emit(...));
  });

  **GOOD**
  on<Event>((event, emit) async {
    await future.whenComplete(() => emit(...));
  });
||#0      _AssertionError._doThrowNew (dart:core-patch/errors_patch.dart:51:61)
#1      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:40:5)
#2      _Emitter.call (package:bloc/src/emitter.dart:114:7)
#3      BleInputBloc._onOpenBleInput.<anonymous closure> (package:blood_pressure_app/components/ble_input/ble_input_bloc.dart:68:15)
#4      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#5      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#6      _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#7      _SyncStreamControllerDispatch._sendData (dart:async/stream_controller.dart:784:19)
#8      _StreamController._add (dart:async/stream_controller.dart:658:7)
#9      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#10     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#11     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#12     _SyncStreamControllerDispatch._sendData (dart:async/stream_controller.dart:784:19)
#13     _StreamController._add (dart:async/stream_controller.dart:658:7)
#14     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#15     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#16     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#17     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#18     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
#19     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#20     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#21     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#22     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
#23     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
#24     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
#25     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
#26     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
}}
```

NobodyForNothing, 2024-04-13:
Sorry, little oversight on my part, please try again

andreassch, 2024-04-14:
I still get the same error. I have pulled your commit "fix unawaited" and made a clean and a build, transferred the resulting apk to my phone and installed it. The app shows version number 1034.

NobodyForNothing, 2024-04-15:
Can you try again?

andreassch, 2024-04-19:
Thanks for your fixes. Now it detects my device and shows it. After I click on it, the pane shortly shows a BloodPressureRecord as demonstrated in the screenshot: 
[image]
After about a second, it changes to
[image]
The debug log is:
```
STATE:BleInputClosed
STATE:BleInputLoadInProgress
STATE:BleInputLoadInProgress
STATE:BleInputLoadSuccess
BleInputLoadSuccess:[DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -58, connectable: Connectable.available)]
STATE:BleConnectInProgress
STATE:BleConnectInProgress
STATE:BleConnectSuccess
STATE:BleMeasurementSuccess
STATE:BleConnectFailed
STATE:BleInputLoadInProgress
```

NobodyForNothing, 2024-04-19:
Thanks, that looks like progress! The year is wrong and all other values are null, but it's going in the right direction. I made some changes that _should_ prevent the state from quickly switching to error and added even more logs. Could you try again and check whether that worked? When your measurement device does/shows something, please report that as well.

I'm also tending towards replacing the device selection with logic that selects the first valid device as this would make the measurement process easier (save 1 tap after waiting) and reduce complexity (=> more robust & smaller app). Please feel free to express any thoughts and objections.

andreassch, 2024-04-19:
Thanks for the updates. Now the received measurement stays on the pane. The BloodPressureRecord on the pane shows nonsense values: (59399-04-19 21:31:38.000, 0, 0, [large negative number], , null)
The log is:
```
2024-04-19 21:30:39.370591 - STATE:BleInputClosed
2024-04-19 21:34:06.912582 - STATE:BleInputLoadInProgress
2024-04-19 21:34:07.233795 - STATE:BleInputLoadInProgress
2024-04-19 21:34:07.399958 - STATE:BleInputLoadFailure
2024-04-19 21:34:21.216711 - STATE:BleInputLoadInProgress
2024-04-19 21:34:21.255609 - STATE:BleInputLoadInProgress
2024-04-19 21:35:11.154776 - STATE:BleInputLoadSuccess
BleInputLoadSuccess:[DiscoveredDevice(id: xx:xx:xx:xx:xx:xx, name: boso medicus CE6674 , serviceData: {}, serviceUuids: [00001810-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -59, connectable: Connectable.available)]
2024-04-19 21:35:14.211310 - STATE:BleConnectInProgress
2024-04-19 21:35:14.309607 - STATE:BleConnectInProgress
2024-04-19 21:35:14.488335 - STATE:BleConnectSuccess
BLE MESSAGE: [11, 111, 0, 22, 0, 122, 0, 222, 3, 4, 11, 22, 33, 44, 55, 0, 0, 0]
2024-04-19 21:35:15.301276 - STATE:BleMeasurementSuccess
```
I have changed the values inside the BLE message for data protection reasons. The second value was the measured systolic value, the fourth the measured diastolic value, and the 15th (i.e. the last before the three zeros) the measured pulse. My devices activates Bluetooth after a measurement and simultaneously shows the measured values on its screen.

Regarding the device selection, I suggest to automatically select the device that is bonded to the phone from the list of available devices. Only if two or more bonded devices are available (which should not happen often), I would present a list to the user to choose.

NobodyForNothing, 2024-05-17:
@andreassch I finished integrating the new feature. Could you check that everything works?

If the app still shows the wrong numbers, binary data should now be available in logcat logs. In that case, I would need that data and the matching values displayed on your device. As the numbers aren't just the raw measured values you can't simply modify the numbers before sending. It would be great if you could somehow falsify the measured value during measurement (to protect your data).

Unfortunately, official GATT test data is only available for SIG members.

andreassch, 2024-05-17:
Thanks for your work. Unfortunately, it does not work. Once a measurement is done, the BLE pane gets empty similar as reported in [my comment from 10 Apr](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/issues/80#issuecomment-2048328586), i.e. the app does not show any numbers. The 'copy debug' button is no longer present. When I tried the BLE feature a second time after that, I got the following error:
'package:blood_pressure_app/bluetooth/device_scan_cubit.dart': Failed assertion: line 30 pos 12: '!_flutterBluePlus.isScanningNow': is not true

NobodyForNothing, 2024-05-18:
I pushed something that should help with the failed assertion. 

Not sure about what happened in the first part. Can you tell me what your actions were and what the app shows until it fails?

The copy debug is now removed on purpose as I replaced it with normal logcat logs.

andreassch, 2024-05-19:
The device activates BLE after it performed a measurement. Thus, I first start the new data point dialog and the BLE input in the app, then I perform a measurement with the device, and when it has finished the measurement (and announces it by BLE) the app recognizes it and tries to read it (the waiting circle disappears). The BLE pane in the app turns black and stays like this.

In the logcat I see this:
```
05-19 10:34:29.659 23336 23387 D BluetoothRemoteDevices: Added new device property
05-19 10:34:29.660 23336 23387 D BluetoothRemoteDevices: Property type: 1
05-19 10:34:29.661 23336 23387 D BluetoothRemoteDevices: Remote Device name is: boso medicus CE6674 
05-19 10:34:29.661 23336 23387 D BluetoothRemoteDevices: Property type: 4
05-19 10:34:29.661 23336 23387 D BluetoothRemoteDevices: Skip class update for xx:xx:xx:xx:xx:xx
05-19 10:34:29.661 23336 23387 D BluetoothRemoteDevices: Property type: 5
05-19 10:34:29.661 23336 23387 D BluetoothRemoteDevices: BT_PROPERTY_TYPE_OF_DEVICE xx:xx:xx:xx:xx:xx
05-19 10:34:29.702  5035  5108 I flutter : TRACE: _onScanResult devices: [ScanResult{device: BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: null}, advertisementData: AdvertisementData{advName: boso medicus CE6674 , txPowerLevel: 0, appearance: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [1810]}, rssi: -69, timeStamp: 2024-05-19 10:34:29.699308}]
05-19 10:34:30.098  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-19 10:34:30.098  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
05-19 10:34:30.111  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-19 10:34:30.111  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
05-19 10:34:30.111  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-19 10:34:30.410  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-19 10:34:30.410  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
05-19 10:34:30.416  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-19 10:34:30.416  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
05-19 10:34:30.416  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-19 10:34:30.481  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-19 10:34:30.482  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
05-19 10:34:30.497  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-19 10:34:30.497  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
05-19 10:34:30.497  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-19 10:34:30.694  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-19 10:34:30.694  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
05-19 10:34:30.707  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-19 10:34:30.707  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
05-19 10:34:30.707  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-19 10:34:30.884  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-19 10:34:30.884  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
05-19 10:34:30.892  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-19 10:34:30.892  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
05-19 10:34:30.892  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-19 10:34:31.387  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-19 10:34:31.387  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
05-19 10:34:31.398  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-19 10:34:31.398  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
05-19 10:34:31.398  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-19 10:34:31.543  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-19 10:34:31.544  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
05-19 10:34:31.551  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-19 10:34:31.551  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
05-19 10:34:31.551  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-19 10:34:31.926  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-19 10:34:31.927  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
05-19 10:34:31.971  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-19 10:34:31.971  1219 23482 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
05-19 10:34:31.971  1219 23482 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
```
Is this the part you are interested in?

NobodyForNothing, 2024-05-19:
Thanks, I think I found it. Can you try again?

andreassch, 2024-05-19:
Now I get further: the app shows my device and a connect button. When I click it, I get a connection error. The logcat shows the following flutter entries:
```
05-19 21:37:39.502  7375  7405 I flutter : TRACE: _onScanResult devices: [ScanResult{device: BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: null}, advertisementData: AdvertisementData{advName: boso medicus CE6674 , txPowerLevel: 0, appearance: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [1810]}, rssi: -70, timeStamp: 2024-05-19 21:37:39.499464}]
05-19 21:37:39.510  7375  7405 I flutter : TRACE: _BluetoothInputState _deviceScanCubit: Instance of 'SingleDeviceAvailable'
05-19 21:37:44.530  7375  7405 I flutter : TRACE: _BluetoothInputState _deviceScanCubit: Instance of 'DeviceListLoading'
05-19 21:37:44.582  7375  7405 I flutter : TRACE: _onScanResult devices: [ScanResult{device: BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: null}, advertisementData: AdvertisementData{advName: boso medicus CE6674 , txPowerLevel: 0, appearance: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [1810]}, rssi: -70, timeStamp: 2024-05-19 21:37:39.499464}]
05-19 21:37:44.624  7375  7405 I flutter : TRACE: _onScanResult devices: []
05-19 21:37:44.624  7375  7405 I flutter : TRACE: _onScanResult devices: []
05-19 21:37:44.632  7375  7405 I flutter : TRACE: _onScanResult devices: [ScanResult{device: BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: null}, advertisementData: AdvertisementData{advName: boso medicus CE6674 , txPowerLevel: 0, appearance: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [1810]}, rssi: -73, timeStamp: 2024-05-19 21:37:44.632420}]
05-19 21:37:44.634  7375  7405 I flutter : TRACE: _onScanResult devices: [ScanResult{device: BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: null}, advertisementData: AdvertisementData{advName: boso medicus CE6674 , txPowerLevel: 0, appearance: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [1810]}, rssi: -73, timeStamp: 2024-05-19 21:37:44.632420}]
05-19 21:37:44.637  7375  7405 I flutter : TRACE: _BluetoothInputState _deviceScanCubit: Instance of 'DeviceSelected'
05-19 21:37:44.639  7375  7405 I flutter : TRACE: _startRead
05-19 21:37:44.642  7375  7405 I flutter : TRACE: _BluetoothInputState BleReadCubit: Instance of 'BleReadInProgress'
05-19 21:37:44.662  7375  7405 I flutter : -----------------------------
05-19 21:37:44.662  7375  7405 I flutter : ERROR service discovery:
05-19 21:37:44.680  7375  7405 I flutter : #0      debugPrintStack (package:flutter/src/foundation/assertions.dart:1201:29)
05-19 21:37:44.680  7375  7405 I flutter : #1      Log.err (package:blood_pressure_app/logging.dart:13:7)
05-19 21:37:44.680  7375  7405 I flutter : #2      BleReadCubit._startRead (package:blood_pressure_app/bluetooth/ble_read_cubit.dart:54:11)
05-19 21:37:44.680  7375  7405 I flutter : <asynchronous suspension>
05-19 21:37:44.680  7375  7405 I flutter : BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: null}
05-19 21:37:44.680  7375  7405 I flutter : FlutterBluePlusException | discoverServices | fbp-code: 6 | device is not connected
05-19 21:37:44.705  7375  7405 I flutter : TRACE: _BluetoothInputState BleReadCubit: Instance of 'BleReadFailure'
```

NobodyForNothing, 2024-05-20:
Thanks. I added some code to ensure connection before reading. Could you try again?

If that does work, it would be great to see if it still works when the reconnection block `app/lib/blutooth/ble_read_cubit.dart`, from `if (_device.isDisconnected) {` (l. 48) to `assert(_device.isConnected);` is moved to line 91 (above `late final List<int> data;`)

andreassch, 2024-05-20:
Now the BLE pane stays in the waiting state (with the running circle) forever. The log shows:
```
05-20 16:26:47.190 15361 15386 D BluetoothRemoteDevices: Added new device property
05-20 16:26:47.191 15361 15386 D BluetoothRemoteDevices: Property type: 1
05-20 16:26:47.192 15361 15386 D BluetoothRemoteDevices: Remote Device name is: boso medicus CE6674 
05-20 16:26:47.192 15361 15386 D BluetoothRemoteDevices: Property type: 4
05-20 16:26:47.192 15361 15386 D BluetoothRemoteDevices: Skip class update for xx:xx:xx:xx:xx:xx
05-20 16:26:47.192 15361 15386 D BluetoothRemoteDevices: Property type: 5
05-20 16:26:47.192 15361 15386 D BluetoothRemoteDevices: BT_PROPERTY_TYPE_OF_DEVICE xx:xx:xx:xx:xx:xx
05-20 16:26:47.233 15037 15064 I flutter : TRACE: _onScanResult devices: [ScanResult{device: BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: null}, advertisementData: AdvertisementData{advName: boso medicus CE6674 , txPowerLevel: 0, appearance: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [1810]}, rssi: -71, timeStamp: 2024-05-20 16:26:47.230139}]
05-20 16:26:47.234 15037 15064 I flutter : TRACE: _BluetoothInputState _deviceScanCubit: Instance of 'DeviceSelected'
05-20 16:26:47.236 15037 15064 I flutter : TRACE: _startRead
05-20 16:26:47.236 15037 15064 I flutter : TRACE: BleReadCubit _startRead: Attempting to connect with boso medicus CE6674 
05-20 16:26:47.239 15037 15064 I flutter : TRACE: _BluetoothInputState BleReadCubit: Instance of 'BleReadInProgress'
05-20 16:26:47.252 15037 15037 D [FBP-Android]: [FBP] onMethodCall: connect
05-20 16:26:47.254 15037 15037 D BluetoothGatt: connect() - device: xx:xx:xx:xx:xx:xx, auto: false, eattSupport: false
05-20 16:26:47.255 15037 15037 D BluetoothGatt: registerApp()
05-20 16:26:47.255 15037 15037 D BluetoothGatt: registerApp() - UUID=6ebd9177-02aa-4c25-a114-83b5c8fd96e0
05-20 16:26:47.256 15361 15374 D BtGatt.GattService: registerClient() - UUID=6ebd9177-02aa-4c25-a114-83b5c8fd96e0
05-20 16:26:47.256 15361 15443 I bt_stack: [INFO:gatt_api.cc(1279)] GATT_Register 59e299dd-cc8a-4024-af75-d3cd24a4386e
05-20 16:26:47.256 15361 15443 I bt_stack: [INFO:gatt_api.cc(1300)] allocated gatt_if=9
05-20 16:26:47.257 15361 15386 D BtGatt.GattService: onClientRegistered() - UUID=6ebd9177-02aa-4c25-a114-83b5c8fd96e0, clientIf=9
05-20 16:26:47.257 15037 15053 D BluetoothGatt: onClientRegistered() - status=0 clientIf=9
05-20 16:26:47.258 15361 15374 D BtGatt.GattService: clientConnect() - address=xx:xx:xx:xx:xx:xx, isDirect=true, opportunistic=false, phy=1
05-20 16:26:47.258 15361 15386 D bt_btif_config: btif_get_address_type: Device [xx:xx:xx:xx:xx:xx] address type 0
05-20 16:26:47.258 15361 15386 D bt_btif_config: btif_get_device_type: Device [xx:xx:xx:xx:xx:xx] type 2
05-20 16:26:47.258 15361 15443 I bt_stack: [INFO:gatt_api.cc(1469)] GATT_Connect gatt_if=9, address=xx:xx:xx:xx:xx:xx is_direct 1
05-20 16:26:47.258 15361 15443 I bt_stack: [INFO:gatt_attr.cc(1270)] gatt_sr_init_cl_status: bda=xx:xx:xx:xx:xx:xx, cl_supp_feat=0000, aware=1
05-20 16:26:47.258  1219  1219 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Writing IBS_WAKE_IND
05-20 16:26:47.258  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_ACK: 0xFC
05-20 16:26:47.259  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Signal wack_cond_
05-20 16:26:47.259  1219  1219 D vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Unblocked from waiting for FC, pthread_cond_timedwait ret = 0
05-20 16:26:47.274 15361 15443 I bt_stack: [INFO:btm_ble_connection_establishment.cc(202)] btm_ble_conn_complete BD addr xx:xx:xx:xx:xx:xx Not resolved 
05-20 16:26:47.274 15361 15443 I bt_stack: [INFO:connection_manager.cc(280)] lower the scan parameters used for connecting
05-20 16:26:47.274 15361 15443 I bt_stack: [INFO:btm_ble_bgconn.cc(433)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
05-20 16:26:47.274 15361 15443 W bt_btm  : btm_acl_created hci_handle=2 link_role=0  transport=2
05-20 16:26:47.274 15361 15443 I bt_stack: [INFO:btm_ble_bgconn.cc(433)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted

05-20 16:26:52.308 15361 15443 I bt_stack: [INFO:gatt_attr.cc(567)] gatt_connect_cback: remove untrusted client status, bda=xx:xx:xx:xx:xx:xx
05-20 16:26:52.308 15361 15443 W bt_stack: [WARNING:bta_gattc_act.cc(1372)] bta_gattc_conn_cback: cif=3 connected=0 conn_id=0x0003 reason=0x0008
05-20 16:26:52.308 15361 15443 W bt_stack: [WARNING:bta_gattc_act.cc(1372)] bta_gattc_conn_cback: cif=4 connected=0 conn_id=0x0004 reason=0x0008
05-20 16:26:52.309 15361 15443 W bt_stack: [WARNING:bta_gattc_act.cc(1372)] bta_gattc_conn_cback: cif=5 connected=0 conn_id=0x0005 reason=0x0008
05-20 16:26:52.309 15361 15443 W bt_stack: [WARNING:bta_gattc_act.cc(1372)] bta_gattc_conn_cback: cif=6 connected=0 conn_id=0x0006 reason=0x0008
05-20 16:26:52.309 15361 15443 W bt_stack: [WARNING:bta_gattc_act.cc(1372)] bta_gattc_conn_cback: cif=7 connected=0 conn_id=0x0007 reason=0x0008
05-20 16:26:52.309 15361 15443 W bt_stack: [WARNING:bta_gattc_act.cc(1372)] bta_gattc_conn_cback: cif=8 connected=0 conn_id=0x0008 reason=0x0008
05-20 16:26:52.309 15361 15443 W bt_stack: [WARNING:bta_gattc_act.cc(1372)] bta_gattc_conn_cback: cif=9 connected=0 conn_id=0x0009 reason=0x0008
[... cut ...]
05-20 16:26:52.309 15361 15443 W bt_stack: [WARNING:bta_gattc_act.cc(458)] bta_gattc_open_fail: Cannot establish Connection. conn_id=000000. Return GATT_Status(8)
05-20 16:26:52.309 15361 15386 D BtGatt.GattService: onConnected() - clientIf=9, connId=0, address=xx:xx:xx:xx:xx:xx
05-20 16:26:52.310 15037 15054 D BluetoothGatt: onClientConnectionState() - status=8 clientIf=9 device=xx:xx:xx:xx:xx:xx
05-20 16:26:52.311 15037 15054 D BluetoothGatt: close()
05-20 16:26:52.311 15037 15054 D BluetoothGatt: unregisterApp() - mClientIf=9
05-20 16:26:52.312 15361 15374 D BtGatt.GattService: unregisterClient() - clientIf=9
```

NobodyForNothing, 2024-05-20:
I wonder why the 35-second timeout didn't trigger. Anyways, please try now.

andreassch, 2024-05-20:
Not much has changed in the behaviour.
```
05-20 22:44:47.948 15361 15386 D BluetoothRemoteDevices: Property type: 1
05-20 22:44:47.949 15361 15386 D BluetoothRemoteDevices: Skip name update for xx:xx:xx:xx:xx:xx
05-20 22:44:47.949 15361 15386 D BluetoothRemoteDevices: Property type: 4
05-20 22:44:47.949 15361 15386 D BluetoothRemoteDevices: Skip class update for xx:xx:xx:xx:xx:xx
05-20 22:44:47.949 15361 15386 D BluetoothRemoteDevices: Property type: 5
05-20 22:44:47.949 15361 15386 D BluetoothRemoteDevices: BT_PROPERTY_TYPE_OF_DEVICE xx:xx:xx:xx:xx:xx
05-20 22:44:47.987  1149  1181 I flutter : TRACE: _onScanResult devices: [ScanResult{device: BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: null}, advertisementData: AdvertisementData{advName: boso medicus CE6674 , txPowerLevel: 0, appearance: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [1810]}, rssi: -76, timeStamp: 2024-05-20 22:44:47.985205}]
05-20 22:44:47.988  1149  1181 I flutter : TRACE: _BluetoothInputState _deviceScanCubit: Instance of 'DeviceSelected'
05-20 22:44:47.990  1149  1181 I flutter : TRACE: _startRead
05-20 22:44:47.990  1149  1181 I flutter : TRACE: BleReadCubit _startRead: Attempting to connect with boso medicus CE6674 
05-20 22:44:47.993  1149  1181 I flutter : TRACE: _BluetoothInputState BleReadCubit: Instance of 'BleReadInProgress'
05-20 22:44:48.016  1149  1149 D [FBP-Android]: [FBP] onMethodCall: connect
05-20 22:44:48.018  1149  1149 D BluetoothGatt: connect() - device: xx:xx:xx:xx:xx:xx, auto: false, eattSupport: false
05-20 22:44:48.018  1149  1149 D BluetoothGatt: registerApp()
05-20 22:44:48.018  1149  1149 D BluetoothGatt: registerApp() - UUID=8bbfa63b-beae-4236-8dae-c865f28614f6
05-20 22:44:48.019 15361 15374 D BtGatt.GattService: registerClient() - UUID=8bbfa63b-beae-4236-8dae-c865f28614f6
05-20 22:44:48.019 15361 15443 I bt_stack: [INFO:gatt_api.cc(1279)] GATT_Register ac60b97f-a82f-0c9d-024c-d872c1b2f333
05-20 22:44:48.019 15361 15443 I bt_stack: [INFO:gatt_api.cc(1300)] allocated gatt_if=10
05-20 22:44:48.019 15361 15386 D BtGatt.GattService: onClientRegistered() - UUID=8bbfa63b-beae-4236-8dae-c865f28614f6, clientIf=10
05-20 22:44:48.020  1149  1169 D BluetoothGatt: onClientRegistered() - status=0 clientIf=10
05-20 22:44:48.021 15361 15374 D BtGatt.GattService: clientConnect() - address=xx:xx:xx:xx:xx:xx, isDirect=true, opportunistic=false, phy=1
05-20 22:44:48.021 15361 15386 D bt_btif_config: btif_get_address_type: Device [xx:xx:xx:xx:xx:xx] address type 0
05-20 22:44:48.021 15361 15386 D bt_btif_config: btif_get_device_type: Device [xx:xx:xx:xx:xx:xx] type 2
05-20 22:44:48.021 15361 15443 I bt_stack: [INFO:gatt_api.cc(1469)] GATT_Connect gatt_if=10, address=xx:xx:xx:xx:xx:xx is_direct 1
05-20 22:44:48.021 15361 15443 I bt_stack: [INFO:gatt_attr.cc(1270)] gatt_sr_init_cl_status: bda=xx:xx:xx:xx:xx:xx, cl_supp_feat=0000, aware=1
05-20 22:44:48.021  1219  1219 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Writing IBS_WAKE_IND
05-20 22:44:48.022  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_ACK: 0xFC
05-20 22:44:48.022  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Signal wack_cond_
05-20 22:44:48.022  1219  1219 D vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Unblocked from waiting for FC, pthread_cond_timedwait ret = 0
05-20 22:44:48.056 15361 15443 I bt_stack: [INFO:btm_ble_connection_establishment.cc(202)] btm_ble_conn_complete BD addr xx:xx:xx:xx:xx:xx Not resolved 
05-20 22:44:48.056 15361 15443 I bt_stack: [INFO:connection_manager.cc(280)] lower the scan parameters used for connecting
05-20 22:44:48.056 15361 15443 I bt_stack: [INFO:btm_ble_bgconn.cc(433)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
05-20 22:44:48.056 15361 15443 W bt_btm  : btm_acl_created hci_handle=7 link_role=0  transport=2
05-20 22:44:48.056 15361 15443 I bt_stack: [INFO:btm_ble_bgconn.cc(433)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
05-20 22:44:48.097  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-20 22:44:48.148  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-20 22:44:48.148  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-20 22:44:48.415  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-20 22:44:48.417  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-20 22:44:48.417  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-20 22:44:48.636  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-20 22:44:48.638  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-20 22:44:48.638  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-20 22:44:48.679  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-20 22:44:48.687  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-20 22:44:48.687  1219 15440 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
[... cut ...]
05-20 22:44:53.069  1149  1172 D BluetoothGatt: onClientConnectionState() - status=8 clientIf=10 device=xx:xx:xx:xx:xx:xx
05-20 22:44:53.070  1149  1172 D [FBP-Android]: [FBP] onConnectionStateChange:disconnected
05-20 22:44:53.070  1149  1172 D [FBP-Android]: [FBP]   status: LINK_SUPERVISION_TIMEOUT
05-20 22:44:53.070  1149  1172 D BluetoothGatt: close()
05-20 22:44:53.070  1149  1172 D BluetoothGatt: unregisterApp() - mClientIf=10
05-20 22:44:53.071 15361 15374 D BtGatt.GattService: unregisterClient() - clientIf=10
```

NobodyForNothing, 2024-05-21:
Thanks, can you try again? I also added a timeout that triggers after two minutes, so if the fix doesn't work, look out for that.

andreassch, 2024-05-22:
It reported that the measurement failed after some time that felt longer than 2 minutes. The log is:
```
05-22 21:33:27.072 16856 18885 I flutter : TRACE: _onScanResult devices: [ScanResult{device: BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: null}, advertisementData: AdvertisementData{advName: boso medicus CE6674 , txPowerLevel: 0, appearance: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [1810]}, rssi: -59, timeStamp: 2024-05-22 21:33:27.069743}]
05-22 21:33:27.073 16856 18885 I flutter : TRACE: _BluetoothInputState _deviceScanCubit: Instance of 'DeviceSelected'
05-22 21:33:27.075 16856 18885 I flutter : TRACE: _ensureConnection
05-22 21:33:27.076 16856 18885 I flutter : TRACE: BleReadCubit _ensureConnection: Attempting to connect with boso medicus CE6674 
05-22 21:33:27.079 16856 18885 I flutter : TRACE: _BluetoothInputState BleReadCubit: Instance of 'BleReadInProgress'
05-22 21:33:27.079  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-22 21:33:27.079  1219 15092 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
05-22 21:33:27.091 16856 18885 I flutter : TRACE: BleReadCubit _onConnectionStateChanged: BluetoothConnectionState.disconnected
05-22 21:33:27.092 16856 16856 D [FBP-Android]: [FBP] onMethodCall: connect
05-22 21:33:27.093  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-22 21:33:27.094  1219 15092 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
05-22 21:33:27.094  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-22 21:33:27.094 16856 16856 D BluetoothGatt: connect() - device: xx:xx:xx:xx:xx:xx, auto: false, eattSupport: false
05-22 21:33:27.094 16856 16856 D BluetoothGatt: registerApp()
05-22 21:33:27.094 16856 16856 D BluetoothGatt: registerApp() - UUID=3223b23d-55d1-4b6d-abb7-b0dca48c8cfc
05-22 21:33:27.095 15042 15055 D BtGatt.GattService: registerClient() - UUID=3223b23d-55d1-4b6d-abb7-b0dca48c8cfc
05-22 21:33:27.096 15042 15095 I bt_stack: [INFO:gatt_api.cc(1279)] GATT_Register a9517de5-570b-7f9a-5d85-f17914737f62
05-22 21:33:27.096 15042 15095 I bt_stack: [INFO:gatt_api.cc(1300)] allocated gatt_if=10
05-22 21:33:27.096 15042 15069 D BtGatt.GattService: onClientRegistered() - UUID=3223b23d-55d1-4b6d-abb7-b0dca48c8cfc, clientIf=10
05-22 21:33:27.097 16856 16898 D BluetoothGatt: onClientRegistered() - status=0 clientIf=10
05-22 21:33:27.098 15042 15055 D BtGatt.GattService: clientConnect() - address=xx:xx:xx:xx:xx:xx, isDirect=true, opportunistic=false, phy=1
05-22 21:33:27.098 15042 15069 D bt_btif_config: btif_get_address_type: Device [xx:xx:xx:xx:xx:xx] address type 0
05-22 21:33:27.098 15042 15069 D bt_btif_config: btif_get_device_type: Device [xx:xx:xx:xx:xx:xx] type 2
05-22 21:33:27.098 15042 15095 I bt_stack: [INFO:gatt_api.cc(1469)] GATT_Connect gatt_if=10, address=xx:xx:xx:xx:xx:xx is_direct 1
05-22 21:33:27.098 15042 15095 I bt_stack: [INFO:gatt_attr.cc(1270)] gatt_sr_init_cl_status: bda=xx:xx:xx:xx:xx:xx, cl_supp_feat=0000, aware=1
05-22 21:33:27.098  1219  1219 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Writing IBS_WAKE_IND
05-22 21:33:27.099  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_ACK: 0xFC
05-22 21:33:27.099  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Signal wack_cond_
05-22 21:33:27.099  1219  1219 D vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Unblocked from waiting for FC, pthread_cond_timedwait ret = 0
05-22 21:33:27.122 15042 15095 I bt_stack: [INFO:btm_ble_connection_establishment.cc(202)] btm_ble_conn_complete BD addr xx:xx:xx:xx:xx:xx Not resolved 
05-22 21:33:27.123 15042 15095 I bt_stack: [INFO:connection_manager.cc(280)] lower the scan parameters used for connecting
05-22 21:33:27.123 15042 15095 I bt_stack: [INFO:btm_ble_bgconn.cc(433)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
05-22 21:33:27.123 15042 15095 W bt_btm  : btm_acl_created hci_handle=6 link_role=0  transport=2
05-22 21:33:27.123 15042 15095 I bt_stack: [INFO:btm_ble_bgconn.cc(433)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
05-22 21:33:27.202  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-22 21:33:27.216  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-22 21:33:27.216  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
[...]
05-22 21:33:31.953  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
05-22 21:33:31.987  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
05-22 21:33:31.987  1219 15092 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
05-22 21:33:32.031  1219 15099 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceSleep: TX Awake, Sending SLEEP_IND
05-22 21:33:32.135 15042 15095 E bt_btm  : btm_ble_read_remote_features_complete: failed for handle: 0x0006, status 0x08
05-22 21:33:32.135 15042 15095 I bt_stack: [INFO:btsnoop.cc(652)] clear_l2cap_whitelist: Clearing whitelist from l2cap channel. conn_handle=6 cid=4:4
[... cut ...]
05-22 21:33:32.137 16856 18877 D [FBP-Android]: [FBP] onConnectionStateChange:disconnected
05-22 21:33:32.137 16856 18877 D [FBP-Android]: [FBP]   status: LINK_SUPERVISION_TIMEOUT
05-22 21:33:32.137 16856 18877 D BluetoothGatt: close()
05-22 21:33:32.137 16856 18877 D BluetoothGatt: unregisterApp() - mClientIf=10
05-22 21:33:32.138 15042 15055 D BtGatt.GattService: unregisterClient() - clientIf=10
05-22 21:33:32.148 16856 18885 I flutter : TRACE: BleReadCubit _onConnectionStateChanged: BluetoothConnectionState.disconnected
```

NobodyForNothing, 2024-05-23:
Drat, bluetooth seems to be more complicated than anticipated... 

Thanks for the helpful errors in the trace, future me will appreciate that. I will be on vacation for the next ~2 weeks and can hopefully fix this after that.

NobodyForNothing, 2024-06-07:
I traced the issue back to a permissions error and am now requesting some more permissions. If that works it would be great if you could try removing permissions (from `app/androidapp/src/main/AndroidManigest.xml`, `app/lib/bluetooth/bluetooth_cubit.dart` (l.73-84) and `app/lib/bluetooth/device_scan_cubit.dart` (l. 68)). Please also report how many permission dialogues you see and when they appear.

If that didn't work you can check the android app info for unallowed / missing permissions.

andreassch, 2024-06-07:
When launching the new version, it asks me for location permission. In the settings, all permissions are granted (nearby devices and location). The new measurement dialog shows the message "No bluetooth permissions".

andreassch, 2024-06-09:
The log shows:
```
06-09 20:19:27.995 20443 20443 D permissions_handler: Bluetooth permission missing in manifest
06-09 20:19:27.998 20443 20443 D permissions_handler: Bluetooth permission missing in manifest
06-09 20:19:28.001 20443 20469 I flutter : TRACE: requestPermission: bluetoothPermission = false
```

NobodyForNothing, 2024-06-09:
Could you try again?

NobodyForNothing, 2024-06-10:
I was wondering: Would you be up to go on a call/livechat sometime and get this thing done? The current approach is really exhausting.

andreassch, 2024-06-10:
It still shows the same behaviour on the UI, although the log messages have changed:
```
06-10 21:33:46.597 28592 28592 D permissions_handler: No permissions found in manifest for: []5
06-10 21:33:46.617 28592 28592 D permissions_handler: No permissions found in manifest for: []5
06-10 21:33:46.620 28592 28618 I flutter : TRACE: requestPermission: locationPermission = false
```
In the settings, no location permission is shown any longer (i.e. it seems not to be requested).

@NobodyForNothing We could meet with Jitsi during the weekend to debug together. Let's coordinate via PM.

NobodyForNothing, 2024-06-11:
Did you by any chance ever set a password / something similar on the device?

andreassch, 2024-06-11:
The app gives the error directly when entering the new measurement dialog, i.e. before any communication with the device starts. I even cannot start scanning for devices. You don't need a device to see it. Don't you experience this behaviour? Maybe it is connected to the Android version (I have Android 13).
I can read data from my device with the nRF Connect app.

NobodyForNothing, 2024-06-12:
Yes, I messed up permission requests on android (when I added location requests). The last commit tried to revert the excess permissions (location) after I ruled them out (sorry, I could have tested this).

What I was thinking about in my question was your log from [before](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/issues/80#issuecomment-2125647993). I reread the [docs on status codes](https://pub.dev/packages/flutter_blue_plus#list-of-bluetooth-gatt-errors) and thought there might be some auth in place (I took your answer as a no). Since then I found [other sources](https://github.com/don/cordova-plugin-ble-central/issues/249#issuecomment-222230309) claiming this could be a random disconnect.

NobodyForNothing, 2024-06-16:
Could you check one final time that everything still works as expected?

andreassch, 2024-06-16:
When starting for the first time, it now asks for the permission when selecting Bluetooth input.
After a measurement I can connect to the device, but while trying to read the measurement, an error occurs. The log shows:
```
06-16 20:01:26.805 10048 10078 I flutter : TRACE: BleReadCubit Connection successful
06-16 20:01:26.816 10048 10048 D [FBP-Android]: [FBP] onMethodCall: setNotifyValue
06-16 20:01:26.817 10048 10048 D BluetoothGatt: setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: true
06-16 20:01:26.818 27476 27790 D BtGatt.GattService: registerForNotification() - address=xx:xx:xx:xx:xx:xx enable: true
06-16 20:01:26.818 27476 27537 D BtGatt.GattService: onRegisterForNotifications() - address=null, status=0, registered=1, handle=14
06-16 20:01:26.860 27476 27622 I bt_stack: [INFO:gatt_main.cc(1718)] gatt_data_process op_code = 1, msg_len = 4
06-16 20:01:26.861 27476 27622 I bt_stack: [INFO:gatt_api.cc(943)] GATTC_Discover conn_id=0x0101, disc_type=5, s_handle=0x000e, e_handle=0x000f
06-16 20:01:26.920 27476 27622 I bt_stack: [INFO:gatt_main.cc(1718)] gatt_data_process op_code = 19, msg_len = 0
06-16 20:01:26.921 27476 27622 D bt_bta_gattc: Handling write response type: 2: handle 0x000f
06-16 20:01:26.922 10048 11621 D [FBP-Android]: [FBP] onDescriptorWrite:
06-16 20:01:26.922 10048 11621 D [FBP-Android]: [FBP]   chr: 2a05
06-16 20:01:26.922 10048 11621 D [FBP-Android]: [FBP]   desc: 2902
06-16 20:01:26.922 10048 11621 D [FBP-Android]: [FBP]   status: GATT_SUCCESS (0)
06-16 20:01:26.940 10048 10078 I flutter : TRACE: BleReadCubit allServices: [BluetoothService{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1800, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1800, secondaryServiceUuid: null, characteristicUuid: 2a00, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1800, secondaryServiceUuid: null, characteristicUuid: 2a01, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: A
06-16 20:01:26.944 10048 10078 I flutter : TRACE: BleReadCubit allCharacteristics: [BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1810, secondaryServiceUuid: null, characteristicUuid: 2a35, descriptors: [BluetoothDescriptor{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1810, characteristicUuid: 2a35, descriptorUuid: 2902, lastValue: []}], properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: false, notify: false, indicate: true, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1810, secondaryServiceUuid: null, characteristicUuid: 2a49, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}
06-16 20:01:26.944 10048 10078 I flutter : -----------------------------
06-16 20:01:26.944 10048 10078 I flutter : ERROR no characteristic:
06-16 20:01:26.963 10048 10078 I flutter : #0      debugPrintStack (package:flutter/src/foundation/assertions.dart:1201:29)
06-16 20:01:26.963 10048 10078 I flutter : #1      Log.err (package:blood_pressure_app/logging.dart:13:7)
06-16 20:01:26.963 10048 10078 I flutter : #2      BleReadCubit._onConnectionStateChanged (package:blood_pressure_app/bluetooth/ble_read_cubit.dart:159:11)
06-16 20:01:26.963 10048 10078 I flutter : <asynchronous suspension>
06-16 20:01:26.963 10048 10078 I flutter : BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: Instance of 'BmDiscoverServicesResult'}
06-16 20:01:26.966 10048 10078 I flutter : [BluetoothService{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1800, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1800, secondaryServiceUuid: null, characteristicUuid: 2a00, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1800, secondaryServiceUuid: null, characteristicUuid: 2a01, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 18
```

NobodyForNothing, 2024-06-17:
copy paste error, try again

andreassch, 2024-06-17:
Now I have the following situation: after a measurement, the reading measurement message is shown until the device switches itself off, when it changes to measurement failed. However, the correct values are taken over. When closing the measurement failed window, tapping on Save is first ignored. After changing some content (e.g. removing the entry "notes" inside the notes field), clicking Save works.
The log shows the following exception before the device is found:
```
06-17 18:34:55.648 18390 18416 I flutter : TRACE: _BluetoothInputState _deviceScanCubit: Instance of 'DeviceListLoading'
06-17 18:34:55.713 18390 18416 I flutter : TRACE: _onScanResult devices: []
06-17 18:34:55.715 18390 18416 E flutter : [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: 'package:blood_pressure_app/bluetooth/device_scan_cubit.dart': Failed assertion: line 75 pos 12: '_flutterBluePlus.isScanningNow': is not true.
06-17 18:34:55.715 18390 18416 E flutter : #0      _AssertionError._doThrowNew (dart:core-patch/errors_patch.dart:51:61)
06-17 18:34:55.715 18390 18416 E flutter : #1      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:40:5)
06-17 18:34:55.715 18390 18416 E flutter : #2      DeviceScanCubit._onScanResult (package:blood_pressure_app/bluetooth/device_scan_cubit.dart:75:12)
06-17 18:34:55.715 18390 18416 E flutter : #3      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
06-17 18:34:55.715 18390 18416 E flutter : #4      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:365:11)
06-17 18:34:55.715 18390 18416 E flutter : #5      _DelayedData.perform (dart:async/stream_impl.dart:541:14)
06-17 18:34:55.715 18390 18416 E flutter : #6      _PendingEvents.handleNext (dart:async/stream_impl.dart:646:11)
06-17 18:34:55.715 18390 18416 E flutter : #7      _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:617:7)
06-17 18:34:55.715 18390 18416 E flutter : #8      _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
06-17 18:34:55.715 18390 18416 E flutter : #9      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
06-17 18:34:55.715 18390 18416 E flutter : 
```
Later, it connects and receives data that are decoded to BleMeasurementData with the correct values for systolic, diastolic, pulse, and date. After that, the log reads:
```
06-17 18:35:36.819 18390 18416 I flutter : TRACE: _BluetoothInputState _deviceScanCubit: Instance of 'DeviceSelected'
06-17 18:35:36.820 18390 18416 I flutter : TRACE: _BluetoothInputState BleReadCubit: Instance of 'BleReadInProgress'
[...]
06-17 18:35:37.869 18390 18416 I flutter : TRACE: BleReadCubit _onConnectionStateChanged: BluetoothConnectionState.connected
06-17 18:35:37.869 18390 18416 I flutter : TRACE: BleReadCubit allServices: [BluetoothService{remoteId: yy:yy:yy:yy:yy:yy, serviceUuid: 1800, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: yy:yy:yy:yy:yy:yy, serviceUuid: 1800, secondaryServiceUuid: null, characteristicUuid: 2a00, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: yy:yy:yy:yy:yy:yy, serviceUuid: 1800, secondaryServiceUuid: null, characteristicUuid: 2a01, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: A
06-17 18:35:37.869 18390 18416 I flutter : TRACE: BleReadCubit allCharacteristics: [BluetoothCharacteristic{remoteId: yy:yy:yy:yy:yy:yy, serviceUuid: 1810, secondaryServiceUuid: null, characteristicUuid: 2a35, descriptors: [BluetoothDescriptor{remoteId: yy:yy:yy:yy:yy:yy, serviceUuid: 1810, characteristicUuid: 2a35, descriptorUuid: 2902, lastValue: [2, 0]}], properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: false, notify: false, indicate: true, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: [nn, nnn, n, nn, n, nn, n, nnn, n, n, nn, nn, nn, n, nn, n, n, n]}, BluetoothCharacteristic{remoteId: yy:yy:yy:yy:yy:yy, serviceUuid: 1810, secondaryServiceUuid: null, characteristicUuid: 2a49, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryp
06-17 18:35:37.869 18390 18416 I flutter : TRACE: BleReadCubit indicationsSet: true
```
Here, the measurement seems to be read again. It continues:
```
06-17 18:35:42.125 18390 18416 I flutter : TRACE: BleReadCubit _onConnectionStateChanged: BluetoothConnectionState.disconnected
06-17 18:35:42.125  2478  2615 D BluetoothA2dp: getState(xx:xx:xx:xx:xx:xx)
06-17 18:35:42.126 26815 26815 D BluetoothMapService: onReceive: android.bluetooth.device.action.ACL_DISCONNECTED
06-17 18:35:42.126 26815 26815 I BluetoothMapService: sRemoteDevice :null device:yy:yy:yy:yy:yy:yy
[... cut ...]
06-17 18:36:12.181 18390 18416 E flutter : #7      _PendingEvents.handleNext (dart:async/stream_impl.dart:646:11)
06-17 18:36:12.181 18390 18416 E flutter : #8      _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:617:7)
06-17 18:36:12.181 18390 18416 E flutter : #9      _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
06-17 18:36:12.181 18390 18416 E flutter : #10     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
06-17 18:36:12.181 18390 18416 E flutter : 
```

NobodyForNothing, 2024-06-17:
Could you try runing with the `--release` flag?

NobodyForNothing, 2024-06-17:
and what happens when the cuff is too loose or the measurement position is wrong?

andreassch, 2024-06-18:
With the new version, it reads a measurement successfully and fills in the correct values into the fields in the dialog, however, the reading measurement pane stays endlessly. Tapping Save does not work in this situation.
In the log, I found the following exception (might be unrelated to the behaviour mentioned above):
```
06-18 21:09:48.330 14047 15059 I flutter : TRACE: BleReadCubit _onConnectionStateChanged: BluetoothConnectionState.disconnected
06-18 21:09:48.334  2921  3326 I bt_stack: [INFO:btm_ble_bgconn.cc(433)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
06-18 21:09:48.334 14047 15059 I flutter : -----------------------------
06-18 21:09:48.334 14047 15059 I flutter : ERROR BleReadCubit _device.connect failed::
06-18 21:09:48.352 14047 15059 I flutter : #0      debugPrintStack (package:flutter/src/foundation/assertions.dart:1201:29)
06-18 21:09:48.352 14047 15059 I flutter : #1      Log.err (package:blood_pressure_app/logging.dart:13:7)
06-18 21:09:48.352 14047 15059 I flutter : #2      BleReadCubit._ensureConnection (package:blood_pressure_app/bluetooth/ble_read_cubit.dart:100:13)
06-18 21:09:48.352 14047 15059 I flutter : <asynchronous suspension>
06-18 21:09:48.352 14047 15059 I flutter : #3      BleReadCubit._onConnectionStateChanged (package:blood_pressure_app/bluetooth/ble_read_cubit.dart:126:7)
06-18 21:09:48.352 14047 15059 I flutter : <asynchronous suspension>
06-18 21:09:48.352 14047 15059 I flutter : BluetoothDevice{remoteId: xx:xx:xx:xx:xx:xx, platformName: boso medicus CE6674 , services: Instance of 'BmDiscoverServicesResult'}
06-18 21:09:48.353 14047 15059 I flutter : FlutterBluePlusException | connect | android-code: 255 | UNKNOWN_HCI_ERROR (255)
06-18 21:09:48.353 14047 15059 I flutter : TRACE: BleReadCubit _ensureConnection: Device not connected
```

Trying to run `flutter run --flavor github --release` fails:
```
Caught exception: Already watching path: /opt/flutter/packages/flutter_tools/gradle

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:packageGithubRelease'.
> A failure occurred while executing com.android.build.gradle.tasks.PackageAndroidArtifact$IncrementalSplitterRunnable
   > SigningConfig "release" is missing required property "storeFile".

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 16s
Running Gradle task 'assembleGithubRelease'...                     76,7s
Error: Gradle task assembleGithubRelease failed with exit code 1
#0      throwToolExit (package:flutter_tools/src/base/common.dart:10:3)
#1      AndroidGradleBuilder.buildGradleApp
(package:flutter_tools/src/android/gradle.dart:498:9)
<asynchronous suspension>
#2      AndroidGradleBuilder.buildApk
(package:flutter_tools/src/android/gradle.dart:223:5)
<asynchronous suspension>
#3      AndroidDevice.startApp
(package:flutter_tools/src/android/android_device.dart:583:7)
<asynchronous suspension>
#4      FlutterDevice.runCold
(package:flutter_tools/src/resident_runner.dart:533:33)
<asynchronous suspension>
#5      ColdRunner.run (package:flutter_tools/src/run_cold.dart:55:28)
<asynchronous suspension>
#6      RunCommand.runCommand
(package:flutter_tools/src/commands/run.dart:833:27)
<asynchronous suspension>
#7      FlutterCommand.run.<anonymous closure>
(package:flutter_tools/src/runner/flutter_command.dart:1394:27)
<asynchronous suspension>
#8      AppContext.run.<anonymous closure>
(package:flutter_tools/src/base/context.dart:153:19)
<asynchronous suspension>
#9      CommandRunner.runCommand (package:args/command_runner.dart:212:13)
<asynchronous suspension>
#10     FlutterCommandRunner.runCommand.<anonymous closure>
(package:flutter_tools/src/runner/flutter_command_runner.dart:372:9)
<asynchronous suspension>
#11     AppContext.run.<anonymous closure>
(package:flutter_tools/src/base/context.dart:153:19)
<asynchronous suspension>
#12     FlutterCommandRunner.runCommand
(package:flutter_tools/src/runner/flutter_command_runner.dart:308:5)
<asynchronous suspension>
#13     run.<anonymous closure>.<anonymous closure>
(package:flutter_tools/runner.dart:130:9)
<asynchronous suspension>
#14     AppContext.run.<anonymous closure>
(package:flutter_tools/src/base/context.dart:153:19)
<asynchronous suspension>
#15     main (package:flutter_tools/executable.dart:93:3)
<asynchronous suspension>
```

NobodyForNothing, 2024-06-19:
Ah sorry, forgot about the whole signing thing. You can leave builds with ` --release` out, they were only meant as a quick way to disable assertions anyways.

I fixed the bug you described (and managed to write a unit test for it :)). Could you try again?

andreassch, 2024-06-22:
The behaviour is still the same: the reading measurement pane stays infinitely. I see in the log:
```
I/flutter ( 1582): TRACE: BleReadCubit data received: [n, n, n, n, n, n, n, n, n, n, n, n, n, n, n, n, n, n]
I/flutter ( 1582): TRACE: BleReadCubit decoded: BleMeasurementData{systolic: x, diastolic: x, meanArterialPressure: x, isMMHG: true, pulse: x, userID: null, status: BleMeasurementStatus{bodyMovementDetected: false, cuffTooLose: false, irregularPulseDetected: false, pulseRateInRange: true, pulseRateExceedsUpperLimit: false, pulseRateIsLessThenLowerLimit: false, improperMeasurementPosition: false}, timestamp: 2024-06-22 16:55:58.000}
I/flutter ( 1582): TRACE: _BluetoothInputState _deviceScanCubit: Instance of 'DeviceSelected'
I/flutter ( 1582): TRACE: _BluetoothInputState BleReadCubit: Instance of 'BleReadInProgress'
D/[FBP-Android]( 1582): [FBP] onMethodCall: discoverServices
D/BluetoothGatt( 1582): discoverServices() - device: xx:xx:xx:xx:xx:xx
D/BluetoothGatt( 1582): onSearchComplete() = Device=xx:xx:xx:xx:xx:xx Status=0
D/[FBP-Android]( 1582): [FBP] onServicesDiscovered:
D/[FBP-Android]( 1582): [FBP]   count: 6
D/[FBP-Android]( 1582): [FBP]   status: 0GATT_SUCCESS
D/[FBP-Android]( 1582): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt( 1582): setCharacteristicNotification() - uuid: 00002a35-0000-1000-8000-00805f9b34fb enable: true
D/[FBP-Android]( 1582): [FBP] onDescriptorWrite:
D/[FBP-Android]( 1582): [FBP]   chr: 2a35
D/[FBP-Android]( 1582): [FBP]   desc: 2902
D/[FBP-Android]( 1582): [FBP]   status: GATT_SUCCESS (0)
I/flutter ( 1582): TRACE: BleReadCubit _onConnectionStateChanged: BluetoothConnectionState.connected
I/flutter ( 1582): TRACE: BleReadCubit allServices: [BluetoothService{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1800, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1800, secondaryServiceUuid: null, characteristicUuid: 2a00, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1800, secondaryServiceUuid: null, characteristicUuid: 2a01, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: A
I/flutter ( 1582): TRACE: BleReadCubit allCharacteristics: [BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1810, secondaryServiceUuid: null, characteristicUuid: 2a35, descriptors: [BluetoothDescriptor{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1810, characteristicUuid: 2a35, descriptorUuid: 2902, lastValue: [2, 0]}], properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: false, notify: false, indicate: true, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: [n, n, n, n, n, n, n, n, n, n, n, n, n, n, n, n, n, n]}, BluetoothCharacteristic{remoteId: xx:xx:xx:xx:xx:xx, serviceUuid: 1810, secondaryServiceUuid: null, characteristicUuid: 2a49, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncry
I/flutter ( 1582): TRACE: BleReadCubit indicationsSet: true
D/BluetoothGatt( 1582): onClientConnectionState() - status=19 clientIf=9 device=xx:xx:xx:xx:xx:xx
D/[FBP-Android]( 1582): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android]( 1582): [FBP]   status: REMOTE_USER_TERMINATED_CONNECTION
D/BluetoothGatt( 1582): close()
D/BluetoothGatt( 1582): unregisterApp() - mClientIf=9
I/flutter ( 1582): TRACE: BleReadCubit _onConnectionStateChanged: BluetoothConnectionState.disconnected
I/flutter ( 1582): TRACE: BleReadCubit _onConnectionStateChanged: BluetoothConnectionState.disconnected
I/flutter ( 1582): TRACE: BleReadCubit _onConnectionStateChanged disconnected: DisconnectReason{platform: ErrorPlatform.android, code: 19, REMOTE_USER_TERMINATED_CONNECTION} Attempting reconnect
I/flutter ( 1582): TRACE: BleReadCubit _ensureConnection
I/flutter ( 1582): TRACE: BleReadCubit _ensureConnection: Attempting to connect with boso medicus CE6674 
D/[FBP-Android]( 1582): [FBP] onMethodCall: connect
D/BluetoothGatt( 1582): connect() - device: xx:xx:xx:xx:xx:xx, auto: false, eattSupport: false
D/BluetoothGatt( 1582): registerApp()
D/BluetoothGatt( 1582): registerApp() - UUID=4ada1a42-64ab-4123-b959-2076647af128
D/BluetoothGatt( 1582): onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt( 1582): onClientConnectionState() - status=255 clientIf=9 device=xx:xx:xx:xx:xx:xx
D/[FBP-Android]( 1582): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android]( 1582): [FBP]   status: UNKNOWN_HCI_ERROR (255)
D/BluetoothGatt( 1582): close()
D/BluetoothGatt( 1582): unregisterApp() - mClientIf=9
I/flutter ( 1582): TRACE: BleReadCubit _onConnectionStateChanged: BluetoothConnectionState.disconnected
[... cut ...]
I/flutter ( 1582): TRACE: BleReadCubit _ensureConnection: Attempting to connect with boso medicus CE6674 
D/[FBP-Android]( 1582): [FBP] onMethodCall: connect
D/BluetoothGatt( 1582): connect() - device: xx:xx:xx:xx:xx:xx, auto: false, eattSupport: false
D/BluetoothGatt( 1582): registerApp()
D/BluetoothGatt( 1582): registerApp() - UUID=83b42b9c-34e6-4dd5-9f9f-f7def912a198
D/BluetoothGatt( 1582): onClientRegistered() - status=0 clientIf=9
I/flutter ( 1582): TRACE: BleReadCubit close
D/[FBP-Android]( 1582): [FBP] onMethodCall: stopScan
D/BluetoothAdapter( 1582): isLeEnabled(): ON
```

NobodyForNothing, 2024-06-22:
I've made some more adjustments, please try again.

andreassch, 2024-06-23:
Thanks for your work. Now the app shows _Measurement taken successfully!_ and the mean arterial pressure after receiving a measurement. The correct data are filled into the field. I can directly click on Save to take over the measurement. The feature now works fine.

andreassch, 2024-06-23:
A nice extra would be to synchronize the clock of the device after a measurement. This seems to be done with the Date Time characteristic (UUID 0x2A08). I notice that the time of my device is some minutes off.

NobodyForNothing, 2024-06-23:
Thanks for your persistence in testing this. This greatly improved my understanding in BLE GATT and can be useful to other implementing something similar as well.

> A nice extra would be to synchronize the clock of the device after a measurement. 

Personally I wouldn't expect time sync during after every measurement and I'm not sure if there are devices that behave weirdly after that (I could imagine records on the device becoming out of order). It's a bit out of scope, but I can imagine this to be some feature in the settings. Implementing measurements has been quite exhausting already but I would gladly accept any PRs.

closed 2024-06-23 by NobodyForNothing

andreassch, 2024-06-23:
In the mean time I provoked some errors. When moving during a measurement, the device showes Err in its display while the app gets the following result:
```
I/flutter (22320): TRACE: BleReadCubit data received: [22, 255, 7, 255, 7, 255, 7, 232, 7, 6, 23, 14, 35, 10, 255, 7, 0, 0]
I/flutter (22320): TRACE: BleReadCubit decoded: BleMeasurementData{systolic: 2047.0, diastolic: 2047.0, meanArterialPressure: 2047.0, isMMHG: true, pulse: 2047.0, userID: null, status: BleMeasurementStatus{bodyMovementDetected: false, cuffTooLose: false, irregularPulseDetected: false, pulseRateInRange: true, pulseRateExceedsUpperLimit: false, pulseRateIsLessThenLowerLimit: false, improperMeasurementPosition: false}, timestamp: 2024-06-23 14:35:10.000}
```
I notice that the incorrect measurement flag is not set for some reason. The app takes it as a regular measurement.

When the cuff is too loose, the display of the device shows Err CUF. The received data is:
```
I/flutter (22320): TRACE: BleReadCubit data received: [22, 255, 7, 255, 7, 255, 7, 232, 7, 6, 23, 14, 47, 59, 255, 7, 0, 0]
I/flutter (22320): TRACE: BleReadCubit decoded: BleMeasurementData{systolic: 2047.0, diastolic: 2047.0, meanArterialPressure: 2047.0, isMMHG: true, pulse: 2047.0, userID: null, status: BleMeasurementStatus{bodyMovementDetected: false, cuffTooLose: false, irregularPulseDetected: false, pulseRateInRange: true, pulseRateExceedsUpperLimit: false, pulseRateIsLessThenLowerLimit: false, improperMeasurementPosition: false}, timestamp: 2024-06-23 14:47:59.000}
```
Again, the app takes it as a regular measurement. I'm not sure whether the device is not implementing GATT completely correct or the app is not interpreting it right.

[2 more replies]