derdilla/blood-pressure-monitor-fl #117: Option to open add page on startup

NobodyForNothing, 2023-07-29:
**What you want**
A setting that makes the open the add measurement page on startup. The Home page shouldn't even load in case the setting is active. 
In case this feature would get expanded to include opening the statistics page on startup, it should include a button to open the add measurements page.

**Why you want it**
In most cases, the user wants to just enter values when opening the app. This change would allow for saving an extra click and some load time when opening the app.

uli-on, 2023-08-02:
Great, very useful. I often look at the intents that apps offer. Yours had none then.

uli-on, 2023-08-22:
I thiiiiink I've mixed up intents and activities. I've the app Activity launcher that helps you save starters to the home screen, and I love to see them through in order to find shortcuts to e. g. the text editors of file managers. And at that occasion I saw that Blood Pressure Monitor had no such activity to click right through to the data entering screen.

NobodyForNothing, 2023-08-23:
A setting for opening the add page is implemented. 
I can't really support your activity launcher app, as this would fundamentally change how the app works internally (Flutter doesn't use different activities for different screens).

closed 2023-08-23 by NobodyForNothing