derdilla/blood-pressure-monitor-fl #118: Allow colored "Needle Pin" dots to be added to certain Measurements that would appear in the Graph

amie42, 2023-07-31:
**What you want**
I get a rather complex medication to deal with my blood pressure. I would like do document the different meds and dosages in the Graph by annotating the corresponding data point with a user defined, colored dot. Would be cool if these dots could also be added to comment-only data points.

**Why you want it**
It would make it much easier to visualize the impact of certain meds on the blood pressure. Would be easier in discussions with my doc to adjust the dosages.

NobodyForNothing, 2023-07-31:
Sure, this appears to be a reasonable feature to add. 

I would imagine that you need more than just a colored point. When adding the point, adding additional info like defining labels/tags (you can suggest more) that can be added to the measurement wouldn't be a lot of effort. Could this be helpful? 

I also think that adding a vertical line at the point would be more precise than just a point. If you think a point would be better please feel free to say so.

amie42, 2023-08-01:
Nice to hear you like my idea. 
I think it should be kept simple. So an additional column to your Table accepting an RGB Value (or a couple of RGB Values) should be sufficient. That RGB Value could be set via a color picker or presets to be defined in the configuration. 
Additional info would easily go into the current "comment" column. 
Whether you'd implement that info as a dot or as a vertical line, I'd totally leave up to you. 

Thanks a mill
Amie

NobodyForNothing, 2023-09-12:
Here is the current progress, feel free to leave feedback.
[image]

closed 2023-09-17 by NobodyForNothing