derdilla/blood-pressure-monitor-fl #132: No space between columns

kri164, 2023-08-22:
Hello,

there is no space between values of sys and dia columns.

[image]

uli-on, 2023-08-25:
Just as a temporary attempt: have you tried to revert the date format back to the default one? It uses the short year format 'yy' instead your 'yyyy', which might be enough to give some jotas of space between sys and dia. Also, using dots instead of hyphens could create some additional space.

NobodyForNothing, 2023-08-25:
Sadly it isn't as simple as just changing the width of the text inside a column, as the portion of the screen it takes is fixed. What you can do for now is change your device's font size. 

I haven't come up with a good layout for the columns yet that suits everyone. The 2 ideas I have now are to either:
- Add another setting for changing the column widths
- Putting the note below the values (if unset it will be hidden)

If anyone has a clever idea on how to change the layout or what to do with this large timestamp, please add it over the next few days.

uli-on, 2023-09-07:
Part of the space problem is created by the history curve on the upper half of the main page. If the graph were combined with the statistics page on one page "Evaluation", it would be possible to wrap the measurement line to two lines without much disadvantage when scrolling: 

```
Date   Sys/Dia   Note
Time   Pulse     (Note)
```

Alternative:
The annotation is not displayed, but symbolized by an icon to save space, e.g. the info-i. The actual annotation can be viewed on the detail view page (edit screen). This relocation should be made a user choice by means of a setting in the Preferences.

Alternative: Less space for annotations, but a popup for the complete text that comes up when held for a long time, similar to that of the values in the history curve.

Alternative:
Space-saving, predefinable labels instead of annotations, e.g. `[Sports]` instead of `After coming home from running`

kri164, 2023-09-07:
Did try to modify date format, but not helped.
[image]

uli-on, 2023-09-07:
I like the new layout very much, but would like to suggest to add date headers to deanonymise the entries a little, i.e. each new date might have a date header. 

You asked for the colours of the edit & delete buttons. To my mind, these should be in the font colour in order to not overexcite the user. They have a divergent direction (pencil) or a divergent expanse (trash) and by that stand out sufficiently. There's no need to erect "neon signs".

NobodyForNothing, 2023-09-08:
[image]

closed 2023-09-09 by NobodyForNothing

uli-on, 2023-09-10:
Yes, the reduced colours are less "yelling" and less confusing.