derdilla/blood-pressure-monitor-fl #138: Problems on the export page, beyond #128

uli-on, 2023-08-28:
**What happened**
Here's a compact list of issues observed. Should you prefer separate ones, just yell. 

- The "heads up" text is displayed incorrectly even when the defaults aren't changed 
- The OK button does not close this reminder
- There are no custom parameters for CSV export available
- Endless scrollable page footer in light gray, below the headline switch (when CSV is selected) where I've expected the parameter choser

NobodyForNothing, 2023-08-30:
The last 2 should be fixed in v1.4.4, the other ones might also be a consequence of the bug. I would appreciate an update on this, once you get the new version (F-Droid build cycles have been quite slow recently).

Btw: Most of the important app info for bug reporting can be found when tapping on the app version in settings.

uli-on, 2023-09-01:
> I would appreciate an update on this, once you get the new version

Yes, of course, I 'm so keen on trying out the export functionality that I won't forget to report back. 

> Btw: Most of the important app info for bug reporting can be found when tapping on the app version in settings.

Ah, that's great, but it required patience and a good memory. Do you think it would be possible to tap on the page and have all these information in the clipboard?

NobodyForNothing, 2023-09-02:
Noticed this issue as well, I will add copy / create bug report buttons later. 
For now I would just make a screenshot.

uli-on, 2023-09-05:
OK, I've now tried all issues reported here, and all seem fixed. I've yet to find out why the "heads up" text appeared one time without noticeable reason, but I'm not yet versed enough with all these options to be sure. 

But I'm absolutely happy now with the possibility to quickly hand my data over to the My Heart app! Huuuge thanks!! :))

uli-on, 2023-09-10:
> I've yet to find out why the "heads up" text appeared one time without noticeable reason

It looks like there's two triggers for the heads up check, the bookmark/preset button (though not in both directions, i. e. does not remove an already visible dialogue) as well as the process of opening the ex/import page. Not yet covered seems to be changing the file type.

Also, the  heads up dialogue can't be closed when CSV is selected in combination with the My Heart preset.

(Might find some more, but these are the ones I always came across.)

NobodyForNothing, 2023-09-10:
Thanks for the investigation, I haven't really changed the conditions after reworking the export system. I will make sure to look into it!

closed 2023-09-24 by NobodyForNothing