derdilla/blood-pressure-monitor-fl #156: [BUG] - <Unable to import CVS>

mttrld1989, 2023-09-24:
### Description

As per title, I am unable to import a CVS file. Not just one I modified with new readings, but even one that I just exported from the app. The error is "This can't be imported".

NobodyForNothing, 2023-09-24:
Thank you for your bug report. 
I can't reproduce this problem with the default export configuration. Could you provide your export configuration? The quickest way to do so is to click on the version page at the bottom of the settings page and send a screenshot.

mttrld1989, 2023-09-24:
Dure thing

[image]

NobodyForNothing, 2023-09-24:
Thanks for the information. Have you already tried disabling custom export fields?

I just published version 1.5.2, which adds a more helpful error message in case CSV import fails. It would be great to hear about your issue once the version went through the F-Droid build cycle in about a week (I assume you use F-Droid). If you use Google Play, it should be a bit quicker.

mttrld1989, 2023-09-24:
I have tried yes. And you're correct, I use F-Droid.

I will update this once I have more info from the new version.

mttrld1989, 2023-10-03:
Hi, just to let you know that I have installed the 1.5.2 version and this problem seems to have solve itself.

closed 2023-10-03 by NobodyForNothing