derdilla/blood-pressure-monitor-fl #162: [BUG] - Export/import ignores notes and colours

DizzyWasTaken, 2023-09-28:
### Description

After exporting as csv and then importing back, notes are missing. The exported csv file is also lacking quotes around the note text (the default text delimiter of the app).

Adding quotes manually around the notes field in the csv doesn't change the behaviour. This bug was present in the previous version too.

As of the current one, the colour attribute is also removed when exporting/importing.

### Android version

Android 11

### Reproduction steps

```bash
1. Go to the Export / Import menu in the settings.
2. Leave everything as is and export the file
3. Import the exact file that has just been exported
4. Notes and colours are stripped. (Overwrites the current ones if present and just lacks them otherwise)
```

### App debug info

```bash
App version: 1.5.1
```

NobodyForNothing, 2023-09-28:
Thanks for the report. The problem with exporting colors is known. Version 1.5.2 allows exporting them, and exporting colors will be default in 1.5.3.

I can't verify that exporting and importing notes doesn't work, although quotation marks don't seem to be set around the notes.

Could you verify that notes are really removed after reimporting notes.

DizzyWasTaken, 2023-09-28:
Yup. I did it before submitting the report and just tried it again with more entries. Same behaviour.

I also noticed that now, when I add a new measurement with the + button, they aren't immediately displayed. I have to select "lifetime" or another valid time frame to refresh and see all the additions.

NobodyForNothing, 2023-10-01:
The issues about not showing the newest entries and color missing from export have been fixed.

I can't reproduce the notes issue with the default configuration, so please ensure you are using the default configuration. 
Version 1.5.3 will include a button for exporting your configuration, so I can have a look at it.

DizzyWasTaken, 2023-10-01:
I've updated to version 1.5.2 and the notes are all getting imported now, thank you. Still no quotes in the exported file, although it doesn't impact the import.

The app now crashes if I import around 170 entries though. Should I open a new bug report for this issue?

NobodyForNothing, 2023-10-01:
Great that the problem is resolved! I will keep the issue open as a reminder to look into the missing quotes.

Feel free to open a new issue for the crash. Please include information about your device and the configurations you used for exporting.

closed 2023-10-01 by NobodyForNothing

DizzyWasTaken, 2023-10-03:
I've noticed it crashes only in specific circumstances. I'll refrain from opening a new issue for now if this doesn't happen for the majority of operations.

NobodyForNothing, 2023-10-03:
Please do tell me the circumstances. I try to avoid all crashes, even rare ones.

DizzyWasTaken, 2023-10-03:
Alright. I commented in case you were still interested. I'll open a new issue now.