derdilla/blood-pressure-monitor-fl #179: [FEATURE] - Rewrite statistics

NobodyForNothing, 2023-10-01:
### What you want:

- [ ] redesign the statistics page so it doesn't look so strict and clumsy
- [ ] rewrite statistics page code (testable, new style)
- [ ] add more useful metrics
- [ ] ability to choose own metrics, move them and create your own (this one should be done after #164 and #165 are done as it creates allot of additional complexity)

### Why you want it:

Currently the statistics page is only a small thought and doesn't live up to it's potential. It's neither fast, nor something I like to look at. 

In it's current form it isn't really useful to anyone and people export the measurements to different apps for further analysis.

xandro0777, 2024-03-24:
Blood pressure values are often very different at different times of the day so it should be possible to break down the statistics to consider measurements at similar times of day.. eg morning, noon, evening separately.

lutzarno, 2024-08-13:
I would like to be able to select whether the average value or the median value is used for a distribution graph on the statistics page. 
With the average as separating line values tend to accumulate on the left side of that line.

NobodyForNothing, 2024-08-14:
@lutzarno I'm not quite sure I fully understood your idea. The average text is always displayed at the horizontal center of the distribution. If you're talking about the dotted line, it is also positioned independent from the values and the graph should be symmetric to it.

If this is more about seeing the median value, I could display it bellow the average without needing an option 🙂.

lutzarno, 2024-08-14:
So the average displayed at the horizontal center is actually a mean value? Looks like I mistook the average as just (MAX+MIN)/2. 

I guess the distribution of the values was misleading me. I exported the measurements and calculated their mean and median. The differences are actually minimal (Mean: 120,6/72,8/69,0 Median: 120/73/69).

I still can't understand why the graph looks wrong to me then.

NobodyForNothing, 2024-08-14:
@lutzarno 

<details><summary>Here is how I understand the terms (for reference):</summary>

### Average / Mean:
- `sum of values / number of values`

### Median:
1. sort all values
2. take the value in the center (`values[values.count / 2]`)

</details>

### The graph

> I still can't understand why the graph looks wrong to me then.

If you are using an app version before v1.7.1 there was a calculation error in the graph.

To clarify what behavior I expect (this should probably be explained in the app): The graph in the value distribution counts how often you saved the corresponding value. 

For example: take the list of sys measurements `123, 122, 120, 123` will result in the following bars and relative heights `120(1), 121(0), 122(1), 123(2)`.

NobodyForNothing, 2024-09-14:
I would prefer your other solutions like allowing to ignore the time submitted by the device or automatically updating the clock. Two timestamps would be a little unexpected and would need some explanation.

Does the `lifetime` interval miss any measurements for you?

derdilla, 2026-09-28:
I think I implemented everything actionable in this issue. If I missed something important, or there are concrete, helpful ideas: open a focused issue with a precise explanation of what is wanted.

closed 2026-09-28 by derdilla