derdilla/blood-pressure-monitor-fl #185: [BUG] - <title>

DizzyWasTaken, 2023-10-03:
### Description

App crashes after importing a list of 179 entries after first importing a smaller list of 2 entries. It shows the "imported entries" message and gets unresponsive right after.

All the settings are default ones.

This doesn't happen if:
- the bigger list is imported before the smaller list
- the smaller list is imported after the bigger  one
- the list range is not set as "lifetime" before importing the bigger list

After importing a few tens of times, the crash has a 100% reproduction rate on my device.

After restarting the app everything works fine.

Device model: Xiaomi Mi 9T Pro

### Android version

Android 11

### Reproduction steps

```bash
1. Install app
2. Open it
3. Go into "import" and import a list of two entries
4. Go to the main screen and set the range of entries to display as "lifetime" (can be done even just after opening the app for it to still crash)
5. Go into "import" and import a list of 179 entries
6. App is unresponsive
```

### App debug info

```bash
App version 1.5.2
Default settings
```

NobodyForNothing, 2023-10-09:
Should be fixed once 1.5.4 is released.

closed 2023-10-09 by NobodyForNothing

DizzyWasTaken, 2023-10-10:
Thank you. Looking forward to it.