derdilla/blood-pressure-monitor-fl #227: CSV Import Fails

jemail99, 2023-11-08:
I'm using the latest version 1.5.6 on Android 10.

I want to import my MediLog exported csv data into your app. I created a one line csv to test import as follows:
```
formattedTimestamp,systolic,diastolic,pulse,notes,color
2021-03-29 08:00:00,121,76,56,,
```
Import fails with this file, ne error messages no user feedback at all and no entries imported.

Do I need the timestampUnixMs, column? If yes is the required unix time stamp 10 digits or 13 digits (Can you confirm what is a correctly formatted date for the app?
What is the encoding of the csv file UTF8?
What line endings are required in the csv file, PC, UNIX or CR?
What exactly does your app expect as a valid csv file? I have compared a test export of one data point and my test csv and I can't see why the import should fail.

I have generated 10 digit and 13 digit unix timestamps based on an exported test reading from your app, but neither helps during import.

NobodyForNothing, 2023-11-08:
> Do I need the timestampUnixMs, column?

Yes, that's why your sample is failing. Currently, the only way to restore a date from a CSV file is through `timestampUnixMs`

The required Unix timestamp is 13 digits in milliseconds. Example: `1077625200050`. Remember that the column needs the title `timestampUnixMs` when converting your data.

> What is the encoding of the csv file UTF8?

Yes, the file is in UTF8.

> What line endings are required in the csv file, PC, UNIX, or CR?

The app uses PC line endings (`\r\n`) by default but falls back to the UNIX ones(`\n`) in case of problems during parsing.

> What exactly does your app expect as a valid csv file?

Not every format that is exportable is importable. When you set your export format to time and remove the unix timestamp from export the app shows a warn banner that the resulting CSV file won't be importable.

Just for confirmation here is a valid CSV sample:
```text
timestampUnixMs,systolic,diastolic,pulse,notes
1077625200050,100,82,63,""
1077690940200,104,85,60,""
```

jemail99, 2023-11-08:
I used your valid CSV sample and attempted to import it with out success. I created the CSV file using KDE Kate text editor on openSUSE Tumbleweed.
No error message, nothing on import, no values imported at all.
Should the Customize fields option be used at all in import?
Is there any more debugging I can do?

jemail99, 2023-11-08:
I have tested your sample csv file on another Android 14 device and import is successful.
This looks like an android version issue.

jemail99, 2023-11-08:
OK, I imported my medilog data following your sample csv into the app on Android 13. I then exported it as a fresh csv file then imported the fresh csv file on my Android 10 device with no problems.
Bit of a cludge workaround, but it would be good to know why it your sample csv imports on Android 13 and not Android 10.

NobodyForNothing, 2023-11-08:
That would be really great to know 🤔. Great it works though.

Could you provide me with the app settings of the android 10 device? 

I somewhat doubt it's related to android version, as the file system library the app uses is quite high level and the part of the android api to my knowledge hasn't really changed in those versions.

jemail99, 2023-11-09:
Re app settings where would I fin those?

NobodyForNothing, 2023-11-09:
Just use the "Backup settings" option on the main settings screen in the data section.

NobodyForNothing, 2023-11-15:
Closing this as a rare issue that can't be reproduced.
In case more people experience issues with this or something similar, please open a new issue or contact me in some other way.

closed 2023-11-15 by NobodyForNothing

mapi68, 2023-11-17:
There are double spaces between date and time (this is default after install the app):

    yyyy-MM-dd  HH:mm

NobodyForNothing, 2023-11-17:
Wow, thanks for the catch! I'm surprised it stayed in the app so long. Fixed in 698f0d6