derdilla/blood-pressure-monitor-fl #237: [FEATURE] [Android] - Adaptive Launcher Icon; VectorDrawable Icon; Themed / Monochrome Launcher Icon

sebastien46, 2023-12-05:
### What you want:

### Progress:

- [ ] Adaptive Launcher Icon
- [ ] VectorDrawable Icon
- [ ] Monochrome Icon

I'd like to add these features using a pull request, however, if @NobodyForNothing is willing to provide the original vector file used to create the app's bitmap icon (if one exists), I would greatly appreciate it, as creating vectors from bitmaps can be quite messy. 

[Developer Reference Documentation](https://developer.android.com/develop/ui/views/launch/icon_design_adaptive)

### Why you want it:

- Starting with [TargetSDK 26 / Android 8](https://developer.android.com/about/versions/oreo), devices start to use the app's adaptive icon / drawable in order for it to follow user theming and device preferences:
[image]

-  In combination with the [VectorDrawable Class](https://developer.android.com/reference/android/graphics/drawable/VectorDrawable), apps can also use XML vector graphics to render their app icon, which significantly decreases the icon's file size, as it creates the bitmap cache on the fly.

- Starting with [TargetSDK 33 / Android 13](https://developer.android.com/about/versions/13), devices can request an app icon / drawable **vector**'s monochrome version, which can enable the app to render a user-themed / monochromatic version of it's launcher icon:

[image]

NobodyForNothing, 2023-12-05:
Sounds like a good feature, feel free to open a PR. 
I already made an [svg arcticons icon](https://github.com/Donnnno/Arcticons/pull/1651) some time back, but I leave it up to you if you want to create one closer to the original.

sebastien46, 2023-12-05:
> Sounds like a good feature, feel free to open a PR. I already made an [svg arcticons icon](https://github.com/Donnnno/Arcticons/pull/1651) some time back, but I leave it up to you if you want to create one closer to the original.
> 
> The original icon was created using blender.

Alright, I made a monochrome vector using the original bitmap, it was fortunately high-res & simple enough for inkscape to not freak out 😅
Here's what it looks like:
[image]

Now, I just need, to cut it up into different parts to have different colors.

NobodyForNothing, 2023-12-07:
@sebastien46 The blender icon was very scuffed, so I highly appreciate your efforts to create a vector graphic. 

In case you want to look at the colors I used in Blender (beware of lighting), the file is [here](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/blob/main/fastlane/metadata/android/en-US/images/sources/bloodPressureApp.blend).

sebastien46, 2023-12-07:
> @sebastien46 The blender icon was very scuffed, so I highly appreciate your efforts to create a vector graphic.
>
> In case you want to look at the colors I used in Blender (beware of lighting), the [...]

So, I did this 2 ways:

 

## 1. 
 - Tracing the PNG
 - Cutting up the contour into multiple paths
 - Coloring the paths using eyedropped colors from the shadowed png (no gradient)

Result:
[image]

# 2.
- Exporting `.blend` file as Grease Pencil, original colors (no lighting):

[image]

I'm pretty sure you want the colors that match the rendered version the most, not the base colors, correct @NobodyForNothing ?

NobodyForNothing, 2023-12-08:
@sebastien46 yes, please use something close to the original. 
I leave it up to you to decide whether to keep the gradient.

sebastien46, 2023-12-08:
> @sebastien46 yes, please use something close to the original. I leave it up to you to decide whether to keep the gradient.

I don't think I'm experienced with vectors enough to try and recreate rendered shadows with the naked eye into accurate gradients 😅 
The one I made only with the eyedropped colors looks pretty damn close to the real thing imho.

I won't be able to test whether or not these changes worked since I don't have a flutter environment set up, but I can still make a pr & you could try to see if it does on your side.
Does that work for you @NobodyForNothing ?

NobodyForNothing, 2023-12-09:
> I won't be able to test whether or not these changes worked since I don't have a flutter environment set up, but I can still make a pr & you could try to see if it does on your side.

@sebastien46 go ahead, I will check if it works.

sebastien46, 2023-12-09:
> > I won't be able to test whether or not these changes worked since I don't have a flutter environment set up, but I can still make a pr & you could try to see if it does on your side.
> 
> @sebastien46 go ahead, I will check if it works.

Ok, so I was able to test it by copying the path data to another app that already has adaptive icons, and unfortunately it looks like this at the moment:

[image]
#### _Gaps are clearly visible_

I'll try to correct this first.

sebastien46, 2023-12-09:
> Ok, so I was able to test it by copying the path data to another app that already has adaptive icons, and unfortunately it looks like this at the moment: [...]

I'll probably make it in a separate PR Later.
For now I'll just finish the monochrome icon which is easier to not screw up 😆

sebastien46, 2023-12-09:
Opened #239

NobodyForNothing, 2023-12-23:
Can I have a rough estimate of when you plan on finishing this?

I plan to make a release over the next days and want to know if I should wait for this.

sebastien46, 2023-12-29:
I'm on vacation right now, but I'll finish it when I get back in a few days.

closed 2026-10-18 by derdilla