derdilla/blood-pressure-monitor-fl #240: image icon size wrong

NobodyForNothing, 2023-12-09:
- white background on some launchers
- clipped

[image]

sebastien46, 2023-12-09:
Can you try with this icon @NobodyForNothing ?
I added the proper amount of padding.

[image]

sebastien46, 2023-12-09:
> * white background on some launchers

The reason why it's now white is because I set the background color to `android:white`, because I didn't know which color to put here since your original icon didn't have one.

I tried implementing it without specifying `<background/>`

> * clipped

I used the padded icon above, it works now.
I also removed the `background` field entirely from `ic_launcher`, now it defaults to black when put on the launcher & transparent when moved around.

Note: the background of adaptive icons cannot be transparent, it must be set to a specific drawable / color value.

[image]

NobodyForNothing, 2023-12-09:
@sebastien46 Thanks for the resized icon, it is helpful in case your other vector icon doesn't get included until the next release.

It makes sense that the adaptive icon doesn't have a transparent background. I track this in a separate issue because this was an issue before your PR and further investigation is needed.

closed 2023-12-24 by NobodyForNothing