derdilla/blood-pressure-monitor-fl #264: Import data from gadgetbridge

SnejPro, 2024-01-08:
### What you want:

Gadgetbridge is an open source android app which collects data from fitness trackers without the need for an account or cloud sync.

I'd like to import my heart rate from this app to blood-pressure-monitor

### Why you want it:

- store heart rate and medication in one app
- no manual entering heart rate measurements

JohanvdS, 2024-01-10:
And or a Bluetooth connection using pine time here, so gadgetbridge is fine for this but a BT connection would be great.

NobodyForNothing, 2024-01-10:
Bluetooth capabilities seem to be a much-requested feature and are already tracked in #80. But I doubt this project can maintain all the different devices and functionality anytime soon.

[Gadgetbridge](https://codeberg.org/Freeyourgadget/Gadgetbridge) allows GPX exports and seems to have a [intent api](https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Intent-API#debug-actions-intents) that allows syncing data when enabled. I would prefer to use the API for syncing to avoid manual actions or excessive permissions being required to import the data.

Interacting with Gadgetbridge seems like something that has the potential to add support for a variety of Bluetooth-related features. I'm currently not very familiar with Gadgetbridge and interfacing with other apps in general (and how to protect data from other apps during transmission). In case someone with experience wants to help with this, feel free to comment here.

@SnejPro from your issue I understand that you only want to track pulse values and medicine intakes using this app. I'm not sure how suited this app is for the task (it currently specializes in blood pressure values) and would like to hear your experience with it (in case you have any). 
This could be the opportunity to track more types of values and make input and list a bit more customizable.

SnejPro, 2024-01-10:
> @SnejPro from your issue I understand that you only want to track pulse values and medicine intakes using this app. I'm not sure how suited this app is for the task (it currently specializes in [...]

My main problem is that i don't like the gadgetbridge graphs. And it would be great to "merge" the medications and heart rate data and show them in one graph so you can visualize how medicine affects heart rate and blood pressure.

It is possible export the complete gadgetbridge database with one click as a sqlite-database. But it is saved in Android/data and therefore must be moved out there to be accessible for other apps.

NobodyForNothing, 2024-01-10:
@SnejPro are you referring to [the activity](https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Intent-API#activity-sync-and-db-export) or [the backup](https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Data-Export-Import-Merging-Processing#user-content-backup-your-data)? 

Importing a SQL database should be relatively simple to support. Do you have some mock data for me to test importing?

SnejPro, 2024-01-10:
> @SnejPro are you referring to [the activity](https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Intent-API#activity-sync-and-db-export) or [the backup](https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Data-Export-Import-Merging-Processing#user-content-backup-your-data)?

> 
> Importing a SQL database should be relatively simple to support. Do you have some mock data for me to test importing?

The backup. I have a correction: you can select the folder where automatic exports will be stored.

Here is an export from my gadgetbridge app. I had a pine time for a few days. These measurements are in the "PINE_TIME_ACTIVITY_SAMPLE" table. Since a few days i have a Fossil hybrid Gen. 6. You can find its measurements under "HYBRID_HRACTIVITY_SAMPLE". I have randomized the values.
[Gadgetbridge.zip](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/files/13890380/Gadgetbridge.zip)

closed 2024-03-07 by NobodyForNothing