derdilla/blood-pressure-monitor-fl #275: Custom Markings "Add Horizontal Line" Nonfunctional

StellarRemnants, 2024-01-16:
### Description

Attempting to add a custom horizontal line (under Settings > Custom Markings) cannot be completed as the OK button seems to instead function as a Back button and the dialog cannot be closed without closing the app.

### Android version

Android 13

### Reproduction steps

```bash
1. Go to Settings
2. Go to Custom Markings
3. Tap "Add line"
4. Select color (any)
5. Enter number in dialog box
6. Press OK
-> OK does not close dialog box, instead, the background window goes back a screen (i.e. on repeated clicks it goes to Settings screen, then the main app screen, then the app closes)
```

### App debug info

```bash
Running latest app release (1.5.7) with default settings.
```

NobodyForNothing, 2024-01-16:
I can not reproduce the issue in the latest app version. Make sure that the number is on the graph (>30) and that there are measurements in the current interval. 

As of today, the latest available app version is 1.6.1 on Google Play & GitHub and 1.6.0 on F-Droid. Please confirm you still have this issue once you upgrade to the latest versions.

StellarRemnants, 2024-01-16:
Ah, I can confirm that in 1.6.0 the bug is no longer present. It looks the the update was so recent that my f-droid app repo hasn't caught up yet. Issue appears to be resolved.