derdilla/blood-pressure-monitor-fl #288: Not able to Update or Reinstall app

Helkriz, 2024-02-11:
### Description

I am using the app to record my blood pressure regularly and it works fine.
my device is **Google Pixel 6a**. 
I was uing app **v1.5.8**
[image]
and whenever I tried to update (f-droid) it didn't saying installer stopped working. Because new version launched I've made backup of data and uninstalled app and tried to install again with no luck. 

### Android version

Android 14

NobodyForNothing, 2024-02-11:
Thanks for the report. Unfortunately, I can not reproduce this on an emulator.

F-Droid hasn't yet finished building the latest version (1.6.2) of the app. The problem might be resolved on its own once the new version is out. 

In case the problem doesn't resolve or you want a solution now, there are a couple of steps you could take:
1. For safety make sure to export all your measurements and settings from the app
2. Check if you have enough free storage on your device and attempt updating
3. Try uninstalling the app and reinstalling through F-Droid, GitHub, or Play Store (Important: switching between these channels is only possible by uninstalling the app, which WILL cause all unexported data to be lost)

I'd be happy to hear how these approaches worked out for you to be able to help others with similar problems in the future. 
Unfortunately, it's tough to pin point installation and upgrade issues to one specific source. I have no control over most aspects like your Android variant or internal aspects of the SDK and Flutter.

Helkriz, 2024-02-11:
Thank you for the response. I've checked it and I'm still having the issue. But it may be the issue with my phone which I am not sure about. If I ever did find it solved I will let you know. 

Please let me know if I've to close the report or not. 

Thank you.

closed 2024-02-11 by Helkriz