derdilla/blood-pressure-monitor-fl #294: Import issues/suggestions

alensiljak, 2024-02-20:
### Description

Hi! Just found the app on F-Droid. I'd like to test it a bit and use my existing data. I'm creating a CSV manually in the format, i.e.

```
Date,Time,Sys,Dia,Pul
2024-02-20,14:20,140,79,59
```

When trying to import this, I get

> There is no column with title "2024-02-24".

This is also a voice for improving import in general, like having some preview (#241) where one could set the columns, select if there is a header, etc.

### Android version

Android 13, LineageOS 20

### Reproduction steps

```bash
as in Description
```

### Screenshots & Settings backup

```bash
No modifications to the defaults after installation.
```

### App debug info

```bash
Version: 1.6.2
Version number: 32
```

NobodyForNothing, 2024-02-20:
I agree that column selection in the import would make this significantly easier for users and I will add this to #241. For me the error message I get when using the data you provided is:

> There is no column with title "Date".

Could it be that don't have the column titles in the CSV you try to import? This will become easier when column selection is implemented. The message should probably tell you more about the actions you can take.

#### Here is something you can do in the meantime:
1. Click on "Manage export columns"
2. Add an exportformat, specifically a time format of the format `yyyy-MM-dd HH:mm` with the title `Date`. This will require you to change the CSV file in a text editor, as multiple columns for the time are not supported due to the amount of complexity that comes with them: remove `,Time` from the headline and replace the first `,` in the rows with a space.
3. To import the other columns you need to either rename their titles in the file: `systolic,diastolic,pulse` or add columns with that title through the menu add exportformat menu.

```
Date,systolic,diastolic,pulse
2024-02-20 14:20,140,79,59
```

For me this correctly imports the entry

NobodyForNothing, 2024-03-15:
Implemented in #300

closed 2024-03-15 by NobodyForNothing