derdilla/blood-pressure-monitor-fl #306: F-Droid Security Warning Version 1.6.3

T-o-b-i-a-s, 2024-03-20:
### Description

Thanks for the nice and free app. When trying to update the App to 1.6.3 in F-Droid, I get the attached warning message. Could you check that?
[image]

I did not check the commits yet, but did your latest changes include any nstructions that could be considered dangerous by Google Play Protect?

### Android version

Android 12

### Reproduction steps

```bash
With automatic Updates enabled in F-Droid, the app did not update automatically. When trying to update manually, the attached message appears.
```

### Screenshots & Settings backup

```bash
![IMG_20240320_070332](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/assets/34947123/6ed9b7b9-f371-4634-98dd-6c05403276bf)
```

pbanj, 2024-03-20:
I get the same thing. It would be nice if the warning actually said what the problem was instead of just a generic message

rLy07, 2024-03-20:
I also got it in the morning when I tried to install it on a new phone. So I installed the older version 1.6.2 but during the day it did auto upgrade itself without any further warning.

NobodyForNothing, 2024-03-21:
There were no recent changes to how the app behaves on the device and the permissions it consumes.

Play protects complain can originate from the different signing key for F-droid and [Google Play Store](https://play.google.com/store/apps/details?id=com.derdilla.bloodPressureApp) (not sure about why now?). 

You can install the Play version (make sqlitedb backup before) or proceed with the fdroid version.

Unfortunately I don't think this is actionable on my end.

pbanj, 2024-03-22:
> There were no recent changes to how the app behaves on the device and the permissions it consumes.
>
> Play protects complain can originate from the different signing key for F-droid and [Google Play [...]

i have apps that have playstore versions and f-droid/github versions and none of them have ever thrown this error. maybe try and contact google and ask them why

pbanj, 2024-03-22:
fairmail, nova video player, all the "simple" apps ex simple gallery.

NobodyForNothing, 2024-03-22:
An [appeal](https://developers.google.com/android/play-protect/warning-dev-guidance#request-appeal) has been requested.

That google doesn't provide a more clear message is really strange, as the warning string you reported doesn't show up on their [own list](https://developers.google.com/android/play-protect/warning-strings).

pbanj, 2024-03-22:
> An [appeal](https://developers.google.com/android/play-protect/warning-dev-guidance#request-appeal) has been requested.
>
> That google doesn't provide a more clear message is really strange, as the [...]

i hate when companies do this crap. all it does is scare non tech savvy people and piss tech people off, and cause issues for devs who cant do much because theres nothing to go off of. like how hard is it to include an actual reason. instead of just being like "no it bad". hopefully they dont just brush you off.

NobodyForNothing, 2024-03-22:
Based on the apk I can't see how Play Protect comes to this conclusion, as common scanning tools like [VirusTotal](https://www.virustotal.com/gui/file/d7f0ff420225aa4687e2abb5cdbc3719e9a840fe028d15cf567982573f2c3edc) report the F-Droid apk as completely fine.

Although if this is a problem with different signing this might be an opportunity to start with F-Droids reproducible builds. Although that would require F-Droid users to delete and reinstall the app.

pbanj, 2024-03-22:
well you have a way to backup and restore. so having to reinstall wouldnt be that bad. better that than google making you look bad

NobodyForNothing, 2024-03-27:
I can reproduce this neither on an emulator nor on my device. I don't know if Google actually fixed this or if it's just being selective so it would be great to hear from users whether Play Protect still complains.

pbanj, 2024-03-27:
Just tried again. No warning this time. So maybe google actually fixed it

T-o-b-i-a-s, 2024-03-28:
In the meantime F-Droid has automatically updated the app to 1.6.3. with no further confirmation or approval from my side. So it seems the Google Play altert has disappeared. 
I think we can close the issue.

closed 2024-03-28 by NobodyForNothing