derdilla/blood-pressure-monitor-fl #309: Allow user to add more fields to be recorded

xandro0777, 2024-03-24:
### What you want:

All kinds of other data could be recorded as well. Weight, oxygen saturation, lab values, medication changes and many other events the user may care about. 

### Why you want it:

Would be useful.

NobodyForNothing, 2024-03-25:
Thank you for showing interest in this, this is an interesting feature to think about. From a technical perspective, this is being worked on in #257, but I'm still unsure whether this would make the UI too complicated.

## Considerations

The basic thought behind the user interface is that at its core the app is a long table with 3 columns of measurements with a one-click form to add values. When activating measurement on launch input is as easy as pressing the number buttons you read and tapping save. 

The problem I have with adding rarely used input types is that the form would get long and confusing and the list can't really display them. Let me explain:

### List

If I were to add a new column to the list the other columns get either too dense or I get 2D scrolling which is harder for users to navigate and doesn't allow seeing all values at a glance. Additionally widening the Row also means having more unfilled fields.

[image]

For notes and timestamps, I currently have a dropdown on the rows that allows viewing them which reportedly leads some users to use the old list format that is more dense and less supported (see #252) and is the idea of adding new columns all over again.

[image]

To integrate medicine intakes I took a different approach, adding list tiles with a medicine icon in between that display the necessary information. I'm also not happy with this solution as it makes values harder to find and disrupts the reading flow of the list. Especially for rare events like lab results, this would be pretty hard to follow.

[image]

Another idea is adding tabs and filters to the top, which would reduce list space and increase the complexity of using the app. (prototype below)

[image]

### Form 

For the form, I have some workable solutions. You could add fields that have to be activated in the settings (like for medicine intakes) which is a bit confusing but could be improved through a setup process on the first launch. Another idea is having a tab bar like the one for adding export columns which would make usage a lot easier although requiring some more interaction to get to the right tab.

<table style="width: 100%; border-collapse: collapse;"><tr>
    <td>[image]</td>
    <td>[image]</td>
</tr></table>

### Exports

Exports are an entirely different issue I haven't touched upon, but adding more columns to the CSV or just exporting the DB should work fine for most cases. One could consider not exporting data types that have no entry.

## "Conclusion"

I might be drastically overthinking this but right now I don't feel like there is a really viable way to communicate this to the user. To my knowledge, there aren't any other apps out there that do a good job at this and I really don't want to worsen the user experience for those who don't need to input additional values.

In case you or anyone else has some ideas about displaying these values feel free to share. Also, consider that I might be assessing the impact my ideas from above have on the average user incorrectly, and tell me if you think differently.

xandro0777, 2024-03-26:
Hi, thanks for your work. Yes, I am not aware of any Open Source App that would do all this, but it would certainly be great to have. So far after trying a few things I am still fully paper based, nothing comes even close. I have some Termux scripts for things like Warfarin and calculating pulse etc out of GadgetBridge data .. and write down the results on paper.

So I fully agree that fast and easy data entry for the most basic data like blood pressure is very important. In fact so important that I might prefer it to be the first view when the app is invoked.. entering data much more often than reading it and most often remember the last few values pretty well anyway so no need to make them visible at the first glance. Still there is additional information even there to be recorded.. did the device (or stethoscope) indicate any arrhythmia? What kind of arrhytmia? I think there is still some spare space for the most important data in this form. Color entry could be made occupy much less space, medication intake as well, note field could expand or popup when somethig is typed in..

As of medication intake entry it might be a consideration to have the user predefine fixed medication/dose combinations. Also while it may be useful in some cases, the great majority of standard blood pressure medications have a bioactivity of 10 hours to several days so entering them with exact time should be the rare exception. Keeping track of medication changes would seem very useful on the other hand. This, like many other possible additional fields would be a one per day or even less frequently used so having to switch to another screen/tab would be no problem.

Some more frequent cases, like weight or temperature in the morning might warrant a special field (if configured so) that is presented for the first data entry of the day.

The lab values, being usually a not very frequent event should also go to a different screen. In fact most of the time people will be probably too lazy to type all that information so it might be a good idea to offer the possibility to import images/pdf files and just add keywords to classify the document. This way, other data like medical reports and such could be conveniently stored.

As of the list view, I realize that not everything can fit into one line. Maybe some more things might and the user could configure which ones he wants there, the rest could be hidden behind a "..." widget. There might also be a top line - daily summary for data that is once a day and exact time is not relevant.
For values/events that are rare it would be possible to show them in an entirely different view where they would be actually much easier to find and browse then in list view although a hint could be shown in the list view or the daily summary.
Data querying, filtering and selecting different combinations of data and stats to be shown would become more important obviously but would not rush to implement this as many people will want to experiment with exported data anyway.

However.. displaying the additional data is by far not as important as entering it as long as it can be exported.

As of exports, I agree that exporting the database would be easiest and acceptable for most users. CSV also has keyword arguments that could be used in some cases but if using CSV it might be preferable to export separate tables for data like lab values or other rare events of any kind. Also JSON is a pretty easy and popular format for data exchange and should fit almost all needs.

No you are not overthinking this, it has some complexity and needs a lot of thinking.

JohnVeness, 2024-04-13:
I'm a mere user of this app, but I think we should be very careful before considering changing it to an "everything to do with medical readings" app. It currently is excellent at recording BP, with its clever way of automatically moving between fields (e.g. knowing when you've typed a full Sys value so moving to Dia automatically). Adding more fields would lose the fluidity, and hiding them behind optional settings seems messy. Plus if you add more readings, users would expect more graphs and calculations of averages and so on. At that point, you might as well just use a spreadsheet.

5FeetUnder, 2024-10-23:
> Yes, I am not aware of any Open Source App that would do all this

you might want to take a look at [MediLog](https://codeberg.org/toz/MediLog). It can record weight, blood pressure, water intake, glucose, temperature and oximetry values and you can add diary entries for logging symptoms etc.
Only caveat is that it is all manual entry. (And the graphs etc. for blood pressure are probably less nice than from Blood Pressure Monitor)

There are however fields regarding blood pressure measurements that I would argue could be beneficial (and that are in the Bluetooth GATT spec as far as I'm aware):
* Irregular Pulse Detection &rarr; indicates whether an irregular pulse was detected
* Body Movement Detection &rarr; indicates whether movement was detected
* Cuff Fit Detection &rarr; indicates whether improper fit of the cuff was detected

Especially recording the irregular Pulse bit could be nice.
The other two metrics could also just stay in the entry form after transmission as a warning. I guess that the detection of these bits is there already, since when I had a measurement with irregular pulse detected, the app showed a warning that movement was detected on the measurement.