derdilla/blood-pressure-monitor-fl #310: App Does not start on Pixel 6 Pro with Android 14

amie42, 2024-03-26:
### Description

Hi, 
I upgraded to 1.6.3 from an unknown, quite old Version using F-Droid. Now the app won't start. All I get is a black screen with your logo. I forcefully killed the app an restarted, didn't help. I restarted the phone, didn't help. Please let me know if I could pass you any logs for debugging.

Best 
Amie

amie42, 2024-03-26:
B

lol, my shortest reply ever!

NobodyForNothing, 2024-03-26:
Huh, that's interesting. I can't reproduce this locally, even when testing with versions as old as 1.4.9 (around that time you had another issue here). 

B is pretty early during app initialization before most of my code is even run.  From my understanding and research of this problem, if you want to keep your data your best bet is probably to wait for the next release, which will be built with a new flutter version.

In case other people experience similar issues, please report them here so I may be able to narrow the issue down further (please report: old version, new version, installation source, and settings if available).

amie42, 2024-03-28:
Thanks, mate! 
I'll observe F-Droid closely and can't wait for the new release. 
Keep up the great work!

Thanks a mill
Amie

NobodyForNothing, 2024-05-14:
@amie42 sorry that took do long. The F-Droid build [should be ready](https://f-droid.org/packages/com.derdilla.bloodPressureApp/).

NobodyForNothing, 2024-07-02:
Closing this since no new reports came in.

closed 2024-07-02 by NobodyForNothing