derdilla/blood-pressure-monitor-fl #316: Move save button lower

JohnVeness, 2024-04-07:
### What you want:

On the data entry screen, please could the save button be moved lower, e.g. a FAB floating just above the on-screen keyboard.

### Why you want it:

Very many thanks for this app, and for making in open-source! I love how easy it is to enter readings, with the way it automatically moves from one field to the next, including using backspace to go to the previous field. The only tiny "papercut" to the data entry is having to reach up to the top of the screen to save, especially if using the phone one-handed. It'd be great if the save button was more within reach. Thanks for reading!

NobodyForNothing, 2024-04-07:
Yeah, that felt awkward to me too. Could you check whether enabling "bottom app bars" in the setting is sufficient? 

If you want this displayed more prominently feel free to make suggestions.

JohnVeness, 2024-04-07:
Ha, wow, I hadn't even noticed that setting! I wonder if the Settings screen is currently too "busy", with everything in one long screen.

In any case, enabling that makes the Save button nicely accessible. Although it looks weird having the close icon at the bottom too.

NobodyForNothing, 2024-06-25:
Closing this as there the requested feature is already part of the app. The way I see it the convenience aspect also applies to the close icon.

closed 2024-06-25 by NobodyForNothing