derdilla/blood-pressure-monitor-fl #318: Split date and time entry

JohnVeness, 2024-04-13:
### What you want:

In the data entry screen, split the date entry from time, i.e. have two different fields that can be clicked on.

### Why you want it:

This is only very minor, but would save me one tap each time! I quite often have to change the time field (e.g. because I took the reading earlier in the day, but didn't enter it into the app until later), but not the date field. At the moment, I click the combined field and always have to immediately click Next to get to the time entry. As I say, not the end of the world, but slightly annoying each time. It would be great if I could skip directly to the time entry.

The only downsides I can see to having two fields is 1) a little more visual clutter (two boxes rather than one) on the data entry screen, and 2) if there are users who regularly edit both date *and* time, they would actually require one extra screen tap.

Presumably you don't have any analytics to be able to tell whether scenario 2 is common or not. My gut feeling is that it wouldn't be common to want to change both date and time, but I could be wrong.

Thanks for reading :)

NobodyForNothing, 2024-04-13:
Thanks for the input. You are correct, I don't have analytics on this. Material 3 also recommends splitting date and time input and I assume they did some research.

From my own app usage and common use cases I had in mind while building there is no real need for combining the inputs. I would even say that changing the date is never required during regular usage. Currently, the time input shows the time in the user-specified format. Unless you have a better idea I would have to ignore the user format here.

About the visual clutter: Two boxes (without two pens) would be fine but I don't know how (new) users would know that those are editable.

NobodyForNothing, 2024-04-13:
This could be influenced by how #309 turns out.

JohnVeness, 2024-04-13:
Thanks for your considered response. I agree two boxes wouldn't be too cluttered. On my screen, there would even be room for two pens. Although, I'm not sure you need the pens - I think it might be more discoverable than you think for users to try clicking on them to edit them, especially if you make sure they look like the other editable inputs. Maybe it would help if you put field titles (i.e. "Date" and "Time") in the top borders of the two boxes.

I'm not sure what you mean about ignoring the user format for time?

NobodyForNothing, 2024-07-03:
@JohnVeness I mocked up 3 designs for an input. What do you think?

<table><tr><td>

[image]

</td><td>

[image]

</td><td>

[image]

</td></tr></table>

I don't quite like either of them so feel free to suggest something different. The first one feels allot like a text input, the third one is quite cluttered and the second one, while very visually clear looks unlike the rest of the form and has too much emphasis in my opinion.

JohnVeness, 2024-07-03:
Thanks for the mockups. Personally I think the first one looks fine. I get what you're saying that they look like text inputs, but I think that's OK as it would encourage people to click in them, and they would soon see that they are specialist inputs. It'd be like the Color input - it looks like a textbox, but you soon learn it isn't.

PR #376 (2024-08-04): Split date and time input fields
closes #318

### New design:

- split up date and time inputs, no more clicking through dialogues
- halved spacing between multiple inputs in one row
- unified color style (height) with other fields

[image]
closed 2024-08-04 by NobodyForNothing