derdilla/blood-pressure-monitor-fl #322: Import Not Working - timestamps

bl4mm, 2024-05-06:
Trying to import a csc but get an error "There is no column that allows restoring a timestamp".  The csv has both date and time fields so not sure why.  Interestingly I tried exporting one record out and then in again and got the same error.

bl4mm, 2024-05-06:
[bp.csv](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/files/15226533/bp.csv)

NobodyForNothing, 2024-05-07:
The problem is that Date and Time are two seperate columns. I [converted the File](https://pastebin.com/XdgAfrQU) to an importable one in standard format.

> [!IMPORTANT]
> @bl4mm You send your blood pressure data **publically**. I send you the file back for convenience and will it delete it in a week.

## General workflow
1. Open in a text editor like vscode and replace the commas with spaces (and remove the comma from the headline)
2. Go to column manager and create a Time format colum with field format `dd MMM yyyy HH:mm`
3. Proceed to import the file. You can select the columns

> [!TIP]
> You can also use a regex to replace the commas in the file. just make sure you match only the correct elements (`20\d\d,\d\d:`) and put the stuff before and after what you want to replace in brackets (capture groups): `(20\d\d),(\d\d:)` in the replace function you can reference these capture groups by index: `$1 $2`

closed 2024-05-07 by NobodyForNothing

bl4mm, 2024-05-07:
@NobodyForNothing Thanks for the quick and helpful reply.  That worked a treat!