derdilla/blood-pressure-monitor-fl #326: Exporting medication data

Nanacho, 2024-06-13:
Thanks for the work on the app so far! I recently started using it since my doctor asked me to start keeping track of my BP, but I couldn't find a way to export medication usage along with the actual readings. Is there currently a way to do that?

NobodyForNothing, 2024-06-13:
Unfortunately not.

When I build the feature it was a rather special case (it's not even in the main db) that didn't integrate with the export mechanism and at the time I didn't see a clear way to export it. This feature is planned nonetheless: The first step of unifying the db is well underway in #257. This will already show it in db export. In a second step intakes will be added to CSV export, either as a optional column or by exporting JSON (lots of boilerplate) or ZIP/TAR archives (harder to work with) as proposed by [xandro0777](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/issues/309#issuecomment-2019177062). **Any further input on this is highly appreciated.** Implementing custom export might take longer as I would implement it in a more decoupled fashion to improve maintainability (#305)

Nonch, 2024-06-14:
Awesome! I'm glad to hear this is planned, and that the ability to record it is there to begin with. I look forward to the ability to export so I have something to show my doctor at some point.

I did look through a database export to see if I could create a spreadsheet from that using a script or something, but I wasn't really able to find the medication entries. I might have to take another look. I may also just be misunderstanding that this is an upcoming feature rather than something already implemented, in which case I will happily wait. This app has already been invaluable to me as it is!

As for exporting to .csv, I think it would be fine to have a column for each medication, defaulting to null or 0, for each timestamp. And if the medication is the only input, to leave the BP and HR readings as null would also be fine. I don't know how that would affect a spreadsheet program's statistics calculations though. Another way to do it would be to add a column for medications in general, and just pile them all in as text:

Suggestion 1:
| Timestamp | SYS | DIA | HR | thinneryl | anemaway |
| --- | --- | --- | --- | --- | --- |
| 1449777600 | null | null | null | 20 | 10 |
| 1436295600 | 120 | 75 | 75 | null | 10 |

Suggestion 2:
| Timestamp | SYS | DIA | HR | Medications |
| --- | --- | --- | --- | --- |
| 1343674800 | null | null | null | thinneryl(20),anemaway(10)|
| 1610136000 | 120 | 75 | 75 | anemaway(10) |

I think I would prefer the first approach, but I would just like something convenient to give my doctor.

Seriously, I cannot thank you enough for your work here; I've been using this app every day. So many other apps I found were riddled with ads and microtransactions, or wanted me to set up an account with something. 

Cheers!

PR #380 (2024-08-07): Allow medicine intake export and import
closes #326

Medicine intakes can now be exported to CSV of the format `MedName1(123.0)|MedName2(456.0)` with the number representing dosis in mg. Import of that data is possible as long as there is a medicine with matching designation.

The alternative approach with columns for each medicine wasn't compatible with the column selection logic.
closed 2024-08-07 by NobodyForNothing