derdilla/blood-pressure-monitor-fl #344: Error while migrating to new data format in Version 1.7.0

skipperTux, 2024-07-04:
### Description

Obviously there is some invalid data in the intakes database; a number is expected, but a non-number has been found. Not sure what went wrong on data input. Any way to recover the old database?
[image]

### Android version

Android 14 (LineageOS 21)

### Reproduction steps

```bash
1. Have some data in the database (with wrong intakes format)
2. F-Droid update from 1.6.4 to 1.7.0
3. Start the App
```

### App debug info

```bash
Error:
Build number:35
-----
Error while migrating intakes to new format:
---
FormatException: Invalid number (at character 1)

^

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
pid: 29570, tid: 29593, name 1.ui
os: android arch: arm64 comp: yes sim: no
build_id: '4dbecf54551a0158e1688d73b199b6ef'
isolate_dso_base: 7149a29000, vm_dso_base: 7149a29000
isolate_instructions: 7149c9f840, vm_instructions: 7149c89000
    #00 abs 0000007149cacec7 virt 0000000000283ec7 _kDartIsolateSnapshotInstructions+0xd687
    #01 abs 0000007149cacdb3 virt 0000000000283db3 _kDartIsolateSnapshotInstructions+0xd573
    #02 abs 0000007149ef8bdf virt 00000000004cfbdf _kDartIsolateSnapshotInstructions+0x25939f
    #03 abs 000000714a09775b virt 000000000066e75b _kDartIsolateSnapshotInstructions+0x3f7f1b
    #04 abs 0000007149e2d6f7 virt 00000000004046f7 _kDartIsolateSnapshotInstructions+0x18deb7
    #05 abs 0000007149ef8a23 virt 00000000004cfa23 _kDartIsolateSnapshotInstructions+0x2591e3
    #06 abs 0000007149ef898b virt 00000000004cf98b _kDartIsolateSnapshotInstructions+0x25914b
    #07 abs 0000007149dce5a3 virt 00000000003a55a3 _kDartIsolateSnapshotInstructions+0x12ed63
    #08 abs 000000714a09758f virt 000000000066e58f _kDartIsolateSnapshotInstructions+0x3f7d4f
    #09 abs 0000007149ca0cff virt 0000000000277cff _kDartIsolateSnapshotInstructions+0x14bf
    #10 abs 0000007149ca0773 virt 0000000000277773 _kDartIsolateSnapshotInstructions+0xf33
    #11 abs 0000007149ef88f3 virt 00000000004cf8f3 _kDartIsolateSnapshotInstructions+0x2590b3
    #12 abs 000000714a1a2eaf virt 0000000000779eaf _kDartIsolateSnapshotInstructions+0x50366f
<asynchronous suspension>
    #13 abs 000000714a19f117 virt 0000000000776117 _kDartIsolateSnapshotInstructions+0x4ff8d7
<asynchronous suspension>
    #14 abs 0000007149ec1df4 virt 0000000000498df4 _kDartIsolateSnapshotInstructions+0x2225b4
<asynchronous suspension>
```

NobodyForNothing, 2024-07-04:
Thank you for your report, sorry that happened. You should be able to scroll down in critical error view and export the database (multiple buttons and files). Or wait until the next release where to resolve this.

NobodyForNothing, 2024-07-04:
de-obfuscated stack trace (for reference):
```
FormatException: Invalid number (at character 1)

^

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
pid: 29570, tid: 29593, name 1.ui
os: android arch: arm64 comp: yes sim: no
build_id: '4dbecf54551a0158e1688d73b199b6ef'
isolate_dso_base: 7149a29000, vm_dso_base: 7149a29000
isolate_instructions: 7149c9f840, vm_instructions: 7149c89000
#0      int._tryParseSmi (third_party/dart/sdk/lib/_internal/vm_shared/lib/integers_patch.dart:37:5)
#1      _StringBase._lastNonWhitespace (third_party/dart/sdk/lib/_internal/vm/lib/string_patch.dart)
#2      PointerEventConverter.expand.<anonymous closure> (/home/derdilla/bin/apps/flutter/packages/flutter/lib/src/gestures/converter.dart:172:26)
#3      getMaterialTranslation (/home/derdilla/bin/apps/flutter/packages/flutter_localizations/lib/src/l10n/generated_material_localizations.dart:45517:14)
#4      MaterialLocalizationEl.timePickerHourModeAnnouncement (third_party/dart/sdk/lib/_internal/vm_shared/lib/compact_hash.dart:515:5)
#5      PointerEventConverter.expand.<anonymous closure> (/home/derdilla/bin/apps/flutter/packages/flutter/lib/src/gestures/converter.dart:146:26)
#6      PointerEventConverter.expand.<anonymous closure> (/home/derdilla/bin/apps/flutter/packages/flutter/lib/src/gestures/converter.dart:146:26)
#7      List.sublist (third_party/dart/sdk/lib/_internal/vm/lib/array.dart:242:3)
#8      getMaterialTranslation (/home/derdilla/bin/apps/flutter/packages/flutter_localizations/lib/src/l10n/generated_material_localizations.dart:45512:10)
#9      Future._propagateToListeners (third_party/dart/sdk/lib/async/future_impl.dart:766:3)
#10     _rootScheduleMicrotask (third_party/dart/sdk/lib/async/zone.dart:1459:39)
#11     PointerEventConverter.expand.<anonymous closure> (/home/derdilla/bin/apps/flutter/packages/flutter/lib/src/gestures/converter.dart:146:26)
#12     max (third_party/dart/sdk/lib/_internal/vm/lib/math_patch.dart)
#13     PdfXrefTable._outputCompressed (/home/derdilla/.pub-cache/hosted/pub.dev/pdf-3.11.0/lib/src/pdf/format/xref.dart:259:21)
<asynchronous suspension>
#14     TypeResolver._resolveInvocation (/home/derdilla/.pub-cache/hosted/pub.dev/sqlparser-0.35.1/lib/src/analysis/types/resolving_visitor.dart:683:12)
<asynchronous suspension>
#15     ListToCsvConverter.convertSingleRow (/home/derdilla/.pub-cache/hosted/pub.dev/csv-6.0.0/lib/list_to_csv_converter.dart:226:15)
<asynchronous suspension>
```

skipperTux, 2024-07-04:
Thanks. I already rescued the old db, config and intakes.
Not sure how to proceed. Rescue new db and import old ones? Downgrade to old version?

NobodyForNothing, 2024-07-04:
The medicine intakes file (custom binary format in v1.6.4) somehow got messed up (a medication id is not decodable), likely due to a bug in a previous version or an really unfortunate medicine id. This format had the potential for unreliability, which is why I switched to a proper database.

The file should contain one intake per line (separated by newlines `\n`). Each intake would be of format: `<medId: int>\x00<unixTimestamp: int>\x00<dosis: double>`. Would you be comfortable looking through the intakes file and sending me a defect intake (so I don't get your medical data)? This would allow me to properly decode these entries.

The good news is that if you exported your data and settings, you can delete the app data and import the old measurement database and settings. To import your intakes you will have to wait for the next version when we figure out how to fix these files.

skipperTux, 2024-07-05:
I have rescued three files, two SQLite files `blood_pressure.db` and `config.db`, and I can open these with my [database](https://github.com/xerial/sqlite-jdbc) [tool](http://www.squirrelsql.org/). The last one `medicine.intakes` is a zero length file and I think that is the issue. I remember I played with this feature, added some content and deleted the content again. 
[image]

Obviously the migration does not expect this file to be empty.

NobodyForNothing, 2024-07-05:
Great, then you can just delete the app data and import config.db through `Settings>Restore` settings and import the blood pressure db through `Settings>Export / Import>IMPORT`.

This means there is no intake format that needs to be fixed, but I will stop the app crashing when there is such an error.

PR #346 (2024-07-05): Stop crashing when converting invalid medicine intakes
fixes #344
closed 2024-07-05 by NobodyForNothing