derdilla/blood-pressure-monitor-fl #352: App always imports the first measurement instead the newest

DerSaxxe, 2024-07-10:
### Description

With Medisana BU550 connect the app imports not the newest measurement. Always the oldest one is saved.

Btw: Thank you for development!

### Android version

Android 11

### Reproduction steps

```bash
1. Delete memory of the BU550 connect
2. Make the first measurement and import it.
3. Make another measurement and try to import.
4. You will see that the first measurement was imported.
```

NobodyForNothing, 2024-07-30:
Unfortunately I have no experience with the BU550 connect but if I understood you correctly you have to manually transmit values. (Please correct me if I'm wrong)

The intended scenario for the Bluetooth input feature was the following:
1. You take a measurement
2. Your measurement device automatically advertises data over Bluetooth
3. You open the app and tap on Bluetooth input
4. The app and the device figure out how to prefill measured data

Could you check whether your devices supports such a protocol? If not I can convert this to a feature request for importing data from Bluetooth.

NobodyForNothing, 2024-07-30:
A similar problem was reported on the `Beurer BM 57` through another channel. After attempting the process described in [my reply above](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/issues/352#issuecomment-2258315305), any further reports on which devices work and which don't are welcome:

### Preliminary compatibility table

Reported working    | Reported broken
--------------------|--------------------
Boso medicus CE6674 | Medisana BU550
Omron x7 smart      | Beurer BM 57

NobodyForNothing, 2024-09-03:
That's a great find! Your approach to the solution is also much simpler and stable than my initial idea of device specific code.

Two remarks:
1. From a glance it seems like the python code uses a notification instead of a indication. As far as I know the difference is just a few bits send to the ble device, but that needs more testing (from the behavior reported here it seems like it's fine for these cheap devices).
2. I still wouldn't wait with filling in the measurements (some cheap devices might not implement connection closing) and replace them in the (presumably) rare case a newer measurement (timestamp) is reported later.

5FeetUnder, 2024-10-14:
[ubpm by LazyT (desktop app)](https://codeberg.org/LazyT/ubpm) handles the multiple measurements quite nicely (at least from how it shows to the user), maybe [their bluetooth-plugin code](https://codeberg.org/LazyT/ubpm/src/branch/master/sources/plugins/vendor/generic/bluetooth) is worth a look. After connecting, it just imports all transmitted measurements though, so no "manual filtering" of sorts as far as I could tell.
It's on a GPLv3 license though, so I don't know how that would affect things (just as a heads-up).

PR #432 (2024-11-11): feat: add bluetooth backend abstraction & support devices with multiple measurements
Resolves #423
Resolves #352

> This is a draft pr for now, need to:
> - [x] review it a bit more (in general and Linux compatibility specifically)
> - [x] check my code comments and
> - [x] at minimum add some tests for measurement_multiple.dart
>

- To resolve 423 I decided to add an abstraction layer for the bluetooth implementation. I wasn't fully sure whether the `bluetooth_low_energy` package will be the best (the maintainer is quite responsive tho) so with this abstraction we should be able to quickly sw [... cut]
closed 2024-11-11 by derdilla