derdilla/blood-pressure-monitor-fl #355: Edit and delete in compact list Mode inop.

Der-Flip, 2024-07-12:
I thought I tag that a question before a bug report:

On 1.7.1 (f-droid and Google play) the edit and delete actions in the compact list are inop. You can swipe and the blue tile +pen or the red tile + bin appear but they don't do anything... Is that intentional, collateral damage or a new bug?

NobodyForNothing, 2024-07-12:
Thanks for the issue. This looks like a bug to me, although maybe only a visual one.

Could you try deleting an entry and then fully closing and reopening the app?

Der-Flip, 2024-07-12:
If I swipe for delete and restart with or without an extra tap on the tile and restart the entry is still there.

(Deleting and editing works perfectly normal in normal layout mode.)

NobodyForNothing, 2024-07-12:
Ok, I can confirm this bug is more than visual.

NobodyForNothing, 2024-07-30:
fixed in #364

closed 2024-07-30 by NobodyForNothing