derdilla/blood-pressure-monitor-fl #419: Add name, DOB, reference number (etc.) to PDF output

JohnVeness, 2024-09-04:
### What you want:

When exporting to PDF, I would like to be able to add some information to the top of it, e.g. my name, date of birth, or a reference number.

### Why you want it:

My doctor needs these details when I give them the PDF. At the moment, I have to print it out and handwrite the details, which is annoying!

JohnVeness, 2024-09-04:
I suggest a freeform text input would be better than trying to split this into individual fields.

NobodyForNothing, 2024-09-04:
That's a very good use case for freeform text input.

What I'm afraid of at the moment is that this simple feature will prompt other wishes for configuration ("I want my text in the header/footer", "Could there be equal spacing when I add a tab", "I want to add my doctors multiline postal-address to the back of the first page"). So ideally you would share the pdf with a dedicated app that does this to avoid a giant, rarely used feature.

Since your case is fairly easy to maintain and is very useful I will consider it some more.