derdilla/blood-pressure-monitor-fl #430: App not starting (black screen)

Der-Flip, 2024-09-10:
### App version

1.8.0 (Playstore)

### Bug Description

I'm doing a ticket for my mom. after the update to version 1.8.0 from the playstore the app won't start anymore. the screen goes black and that's it.

device: Samsung A51 (Android 13 [tp1a.220624.014.A515FXXU8HWI1]; One UI 5.1). I don't have more infos really than that. She tried to reboot and deleting the cache (and another reboot), neither worked.

We can say it was definitely the update. she had the app open today twice already. Google Play updated the app 15:30 and now she cannot open the app anymore.

I don't have access to the device, so I can't do further tests than what I instructed her to do already. If there is anything else you need to know, let me know, I'll try my best to instruct her.

For now: Is there any way to backup the data without opening the app?

NobodyForNothing, 2024-09-10:
I'm afraid that's related to https://github.com/flutter/flutter/issues/154103 (basically a bug in samsungs graphics chips). Unfortunately there isn't much I can do about it right now. As the app can't even load data can't be extracted externally.

There are a few ways to approach this. I will definitely push a update as soon as flutter releases a new stable version. This is the safest option but may take one or two weeks.

Google play app signing doesn't really help, but if your mom needs to export her data right away I can try get her on a custom closed alpha release channel. I have never done this before so I would need to do some more testing to ensure it works before attempting it on the "real" device/data.

If you happen to know a way to root the phone without loosing data and are willing to risk it. I could also help you from there. (strongly *discouraged* but listed as theoretical approach)

If possible I would definitely go for option 1 as that way I'm very certain that no data will be lost. As an immediate fix for other samsung users I will try to hold google play rollout right away!

Der-Flip, 2024-09-10:
Hi, thanks for your speedy reply!

and oof that doesn't sound like a great way to spend your evening and upcoming days O.O

She had a pdf exported today for the past 7 days and last month. so should she need to show it to the doctor she can show the pdf (although her doctor has found a liking in the app and the statistics as well).
I was more worried that an update might not be able to recover functionality and whether there was a way to save the data.
I will most certainly urge her (or just do it myself next time I see her) to activate the export csv after entry now that it is working reliably again. :'D For now, pen and paper will do the trick, I offered to run it as dual app on my phone so she can receive a csv once it's running again, so no need for the trouble with the alpha. Thank you so much for offering though, highly appreciated!

I switched to android from BB10 so i am not experienced in rooting and all that stuff and with it being a company phone, yeah no, not gonna do that :'D

thanks for all the effort!

NobodyForNothing, 2024-09-10:
Great to hear that. Don't worry about my evenings ;), this whole situation was caused because I forgot to back from the flutter main branch after debugging a similar issue on emulators. Backporting is fairly straight forward ([branch](https://github.com/NobodyForNothing/blood-pressure-monitor-fl/tree/v1.8.0-samsung-hotfix) and f-droid should be unaffected.

Now it's (hopefully) just waiting for play console to review my update :)

Thanhngoclitmax, 2024-09-11:
Der-Flip

uli-on, 2024-09-13:
@Der-Flip you could try to instruct her how to swipe out the app from the app switcher. Works for me. 

@NobodyForNothing Will post some more info when i've some more time

uli-on, 2024-09-13:
Here are some screenshots I've taken, many flashes, coloured areas, unexpected transparencies, and feels like graphics card artefacts. And my Fairphone 3 has a Samsung display, too. Android 13 with the current (august 5) security update. bpm 1.8.0 (40) from F-Droid

Swiping bpm out of the app switcher, BTW, is no guarantee for making it work as I had to find out, and I seemed to have different results with different soft-keyboards.

[image]
[image]
[image]
[image]
[image]
[image]

NobodyForNothing, 2024-09-14:
@uli-on I'm quite confident this (original) issue is specific to the google play branch. To debug another (developement-related) issue my desktop was on the flutter main branch when building the app for play-store. f-droid only gets the flutter-version number to build from (I gave them the latest beta at the time of release). 

The next release will be build with the latest stable flutter version, which I hope fixes both issues here.

NobodyForNothing, 2024-09-17:
@Der-Flip the release with the old version is out on play now, I hope that helps.

@uli-on I made a mistake in their build config, so the f-droid build will take a few more days. If the issue persists please create a new issue linking your comment where we can sort out the details.

closed 2024-09-17 by NobodyForNothing

Der-Flip, 2024-09-17:
> I hope that helps

thank you so much. I just had word from her that it works as it should (and apparently just in time as she has an appointment tomorrow, so the timing couldn't be any more perfect)

Danke dir!

uli-on, 2024-09-20:
> the f-droid build will take a few more days

Will the update arrive in f-droid with just the source code zips? Cause currently there's no apk for both the fix and the hotfix2.

NobodyForNothing, 2024-09-20:
@uli-on yes, I don't have reproducible builds set up for f-droid. I'm aware that this might not be fixed for samsung users using gh release/obtainum but this release will stio them from downloading the faulty one regardless

uli-on, 2024-09-25:
I've used the fix now for 1 day or so and it looks very good, no flaws, no black or jagged screens any longer. Thanks very much!