derdilla/blood-pressure-monitor-fl #441: F-Droid build failed

linsui, 2024-09-17:
### App version

v1.8.0-samsung-hotfix

### Bug Description

https://gitlab.com/fdroid/fdroiddata/-/jobs/7840543236 In https://github.com/NobodyForNothing/blood-pressure-monitor-fl/commit/ca89b04f64b79266d432faa1d9daf4133d77e84e the flutter version is downgraded. Maybe it shouldn't?

### Details (Screenshots, Debug Info, Configurations)

<details><summary>Attachments</summary>

</details>

NobodyForNothing, 2024-09-17:
The downgrade is intentional, the old(newer) version showed a black screen on some samsung devices. I downgraded until this is fixed.

linsui, 2024-09-17:
But a newer dart is required so the build failed...

NobodyForNothing, 2024-09-17:
The fdroid dart version depends on the flutter version and got downgraded as well. Since that release a new flutter version was published which still has incompatibility with a library.

linsui, 2024-09-17:
```
The current Dart SDK version is 3.4.4.
Because health_data_store depends on build_runner >=2.4.12 which requires SDK version ^3.5.0-259.0.dev, version solving failed.
You can try one of the following suggestions to make the pubspec resolve:
* Try using the Dart SDK version: 3.5.3. See https://dart.dev/get-dart.
* Consider downgrading your constraint on build_runner: dart pub add dev:build_runner:^2.4.11
```

NobodyForNothing, 2024-09-17:
Thank you for the info. As an fix f-droid could get an independent dart version or I could also downgrade the dependencies in the external package.

linsui, 2024-09-17:
Both are OK. :) If we use an independent dart version then you also need to write the version in the pubspec file.

NobodyForNothing, 2024-09-17:
Pushed a new release with the build_runner downgrade :+1:

commit f37de24c (2024-09-17): Downgrade build runner

fixes #441
closed 2024-09-17 by NobodyForNothing