derdilla/blood-pressure-monitor-fl #466: Custom export didn't include readings from today

JohnVeness, 2024-11-02:
### App version

1.8.1

### Bug Description

I wanted to produce a PDF of my readings for my doctor for the last 10 days, including those that I had made earlier that morning. I found I was unable to include the readings from today, just the previous 9 days.

### Steps to Reproduce

1. Settings
2. Export / Import
3. Date range: Custom
4. Pick start date
5. Pick end date (today)
6. Export format: PDF
7. EXPORT button

I thought maybe I could get it working by choosing an end date in the future, but unfortunately the UI wouldn't let me do that.

derdilla, 2024-11-03:
Thanks for the report. Does the home page show the entries, or can you reproduce it there (if so, this might be linked to #462)?

While this is getting fixed you could probably work around the in-future-date-setting by selecting the range in the past and using the arrow buttons to move around.

JohnVeness, 2024-11-03:
Hi, thanks. No, it's not the same as #462, as the measurements were added to the home page. It was just an export problem. Thanks for the hint about the arrows - that should hopefully workaround the problem for next time (I haven't tried yet, though).

PR #467 (2024-11-04): Fix custom date range ending before midnight
When picking custom time the picker didn't ensure the end of the day (23:59:59) is selected, which resulted in newer measurements not being shown (on list and export). 

This PR:
- ensures the time is set to the end of the day
- adds proper testing to interval pickers custom range selection

closes #466
closed 2024-11-04 by derdilla