derdilla/blood-pressure-monitor-fl #482: Warn area filled without graph line

derdilla, 2024-11-13:
### App version

v1.8.2

### Bug Description

It looks like #468 introduced a new issue someone emailed me about. The warn area can now be filled without a graph above it.

[image]

The graph in general looks a bit fucked up there. It was captured on a android 13 Redmi Note 11S with "Xiaomi HypeOS".

### Steps to Reproduce

The offending 2 measurements are:
- 133 70 70
- 123 70 70

### Details (Screenshots, Debug Info, Configurations)

<details><summary>Attachments</summary>

</details>

derdilla, 2024-11-14:
@Intranox, you are right, a definitely `SafeArea` won't hurt (and this screenshot makes it apparent that one is needed). When fixing this it can be fixed as well (or in another PR, the code isn't necessarily related). The point of this issue is the small red area at the top left.

Intranox, 2024-11-14:
> @Intranox, you are right, a `SafeArea` definitely won't hurt (and this screenshot makes it apparent that one is needed). When fixing this it can be fixed as well (or in another PR, the code isn't necessarily related). The point of this issue is the small red area at the top left.

Sorry but I didn't understand well. I can only say that I have not yet encountered this problem with Android 11 on Oneplus. I would still be grateful if you can implement the SafeArea (having the notch a part of the graph is completely lost)

derdilla, 2024-11-18:
fixed in 60fb035, 52580db

closed 2024-11-18 by derdilla