derdilla/blood-pressure-monitor-fl #491: Entering and exporting data

joeyiii57, 2024-11-22:
### App version

v1.8.1

### Bug Description

1. When I enter my weight into the weight tab it stays in the app.  When I export the data there doesn't seem to be any way to export the weight data.
2.  I have a 5 inch screen.  The three buttons on the right hand side of the screen are so intrusive they almost block the edit icon.  It would be considerably more beneficial for me if the three buttons were moved to a menu on the top or bottom.
3. Thanks. 

### Details (Screenshots, Debug Info, Configurations)

<details><summary>Attachments</summary>

</details>
[image]

derdilla, 2024-11-23:
Thank you for the note about the weight export, I completly forgot about that. I hope to add that with the next release. 

I acknowledge that the actions don't really work with a small screen and would be happy to accept any patches about them. For now you can enable the compact list in the settings or drag up the graph/table header if you need to reach the actions.

derdilla, 2025-01-26:
In compact mode you need to swipe to edit or delete entries. csv/pdf weight export is possible through the weight column in the next release (presumably 1.8.5). The sqliteDB export already works great for porting between devices / as a backup.

PR #516 (2025-01-27): Implement bodyweight exporting
This adds an csv/pdf export option for bodyweight entries. While csv import is now theoretically possible, it is not production ready. Hence no customizable column format string has been added for weight export.

Closes #491.
closed 2025-01-27 by derdilla