derdilla/blood-pressure-monitor-fl #514: Can not restore data

MatejKovacic, 2025-01-24:
### App version

v1.8.2

### Bug Description

I have changed my phone. On the old phone I have version `1.8.2` and on the new phone `1.8.4` (don't know why the difference). Anyway, I went to `Settings` - `Backup settings` and send the file to new phone. On new phone I imported the file, and it says `Sucess: restore settings`.

I close the app, but there is no data.

Also, I tried to import CSV, and it is the same.

### Steps to Reproduce

1. `Settings` - `Backup settings` on old phone
2. Send ZIP file to new phone
3. `Settings` - `Restore settings` on new phone
4. Close the app
5. Open the app

### Details (Screenshots, Debug Info, Configurations)

<details><summary>Attachments</summary>

</details>

derdilla, 2025-01-26:
Thanks for the bug report! My best guess is that the data doesn't get to the app altogether. There have been similar issues exporting data on newer android versions. What is your Android version / Device? Does a file picker even pop up?

You also mention closing and opening the app. I don't assume you see the settings change before closing the app? (they should if everything works)

MatejKovacic, 2025-01-26:
I am using GrapheneOS, which is based on AOSP Android and has the latest version of Android (15).

derdilla, 2025-01-26:
Hi @MatejKovacic I'm having some trouble reproducing your issue. Would you be comfortable sharing your settings privately at `contact@derdilla.com`? Although I was able to rule out the issue I initially expected.

MatejKovacic, 2025-01-26:
No problem. Sent it just now.

derdilla, 2025-01-26:
With the advice I relied to you: Is the problem resolved?

MatejKovacic, 2025-01-27:
Yes! I renamed the file to "bp.db" and import was successful.

Also, there is another thing. In GrapheneOS the system is drawing navigation buttons at the top of the app. So "import/export" buttons are covered and are hard to press. Please see the picture:

[image]

My suggestion would be to move "import/export" buttons a little bit higher.

derdilla, 2025-01-27:
That's great! I hope your issues are resolved now.

I created a new issue for tracking the bar problem. If you have more problems just reopen this one or consider creating a new more precise one.

closed 2025-01-27 by derdilla