derdilla/blood-pressure-monitor-fl #518: Translation

MatejKovacic, 2025-01-27:
I just translated the app, however I have some questions (and I also think that some English strings may have errors). Could you check the translation comments?

Also, when do you plan to release new version?

When you will release new version with translations, I will go through the app and see how the translations actually work and then "repair" them.

derdilla, 2025-01-27:
I left some replies :), and I will make sure to ping you here once the next version releases (next 2 weeks preferably).

derdilla, 2025-02-03:
@MatejKovacic the release is already available via github and google play beta. Production Google Play and f-droid might take a few days.

If you notice anything that would justify a change in design, feel free to suggest it.

closed 2025-02-26 by derdilla