derdilla/blood-pressure-monitor-fl #519: Accrescent instead of F-Droid?

MatejKovacic, 2025-01-28:
### What you want:

Please see this - it seems F-Droid has some serious issues:

- https://x.com/GrapheneOS/status/1883895255142932816?t=s1be0McMhmbcKYsYGeM-ZQ&s=19
- [fdroid/fdroiddata#3110 (comment 1613430404)](https://gitlab.com/fdroid/fdroiddata/-/issues/3110#note_1613430404)

Can I suggest to distribute this app through https://accrescent.app/?
I is an Android app store focused on security, privacy, and usability and it is already recommended app store for GrapheneOS users.

derdilla, 2025-01-28:
I don't disagree that F-droid has had some security and stability issues and I'm open to providing the app through other channels. If anyone wants to bring my app to their store for noncommercial purposes they are free to do so and I'd happy to link them in the README.

With that being said: My app is not a security critical component that needs to get updates as soon as they are available. If F-Droid where to stop working some day users could still export their data and move to a different source. I try to guarantee at least one year of backwards compatibility of all data and exports produced by my app, so missing a few updates isn't a problem.

It could be a lack of understanding on my part, but I see multiple red flags with the service you mentioned.
- developers need a (proprietary) GitHub account to log in
- the app is still in "early alpha" state as they put it
- it requires me as a developer to put in work for every release: While this is similar to google play, I honestly don't think it is worth my time for a store without many locked in users. The [developer docs](https://accrescent.app/docs/guide/getting-started/new-app.html) mention domain verification and building up a store presence. Updates through a CD pipeline are not quite feasible/secure through my current setup. I would need to host my own infrastructure to avoid giving my signing keys to potentially untrustworthy third-parties.
- accrescent also doesn't have the same transparency in its leadership f-droid promises and no established reputation
- While its source code is open I couldn't find the data: you need their app to download
- They have no imprint and no legal entity behind them. The website is copyrighted to a single person.

These are the things that come to mind as a developer and potential user. It seems awkwardly similar to Google Play but without a USP. I've seen projects like that come and go often enough to spend time on it before it's more established. If you want to take it up, I'm happy to help with the domain verification and you can find other assets from the play presence used in this repository. Naturally you will need your own signing key.

*tldr*: You do you. I wait for them to become established.

closed 2025-01-28 by derdilla

derdilla, 2025-01-28:
Looks like this wouldn't even be feasible right now: [image]