derdilla/blood-pressure-monitor-fl #552: Saveing defect after update

Jay-Cee, 2025-04-11:
### App version

v1.8.6

### Bug Description

After update v1.8.5 to v1.8.6 dd. 2025.04.11 the save Button no longer saves the data.
Data input is same as before.
(Using GrapheneOS and until today everything was flawless.)

### Steps to Reproduce

1. Open APP
2. enter syst. value
3. enter diast. value
4. enter heart beat rate
5. hit save
6. values do not appear in list

### Details (Screenshots, Debug Info, Configurations)

<details><summary>Attachments</summary>

</details>

derdilla, 2025-04-11:
Thanks for the report. 

Just to make sure: Is the home screen graph set to a good interval? What happens if you set the graph to lifetime and fully close and reopen the app?

derdilla, 2025-04-11:
<details><summary>Here is what I'm seeing</summary>
<p>

https://github.com/user-attachments/assets/6c22ce03-2cb1-4fb6-83bc-a3f2e7331aea

</p>
</details>

Do you have any settings ect. that might affect this?

hanskuster, 2025-04-20:
doesn't work here either.
Just updated, and suddenly I couldn't save data anymore.
I had to revert to version 1.8.5 to enter further data.

derdilla, 2025-04-22:
@hanskuster Do you use a special android distribution (like GraphenOS)?

And regarding your other bug (#556): How did you "downgrade"? Could this be an effect?

hanskuster, 2025-04-22:
I use a Motorola "moto g 5G" with android 10 and rootet.
My wife had this bug first. I extracted the apk from my phone and updated my phone too. Then I had the same issue.
With the extracted apk (version 1.8.5) I could revert both phones back to version 1.8.5. So it is actually a re-install of the previous version 1.8.5

Jay-Cee, 2025-05-23:
> You can now (1.8.7) enable very verbose logging (leaks entered values) in Settings -> App Info. Could either of you try capturing some logs of this with some test data and then send them here? No data leaves the app until you press copy.

Hello,

just had a 'quick and dirty' trial.
What shall say, works AS disired.

Excellent!

Will leave Loggung in the next days.
Here's the one from the trial...

INFO - 2025-05-23T07:37:06.148149||||"Verbose logging set to true"||{null}
FINER - 2025-05-23T07:37:13.293996||BPM[context.EntryUtils]||"createEntry(null)"||{null}
FINER - 2025-05-23T07:37:13.302802||BPM[Lw]||"Initializing with null"||{null}
FINE - 2025-05-23T07:37:13.302861||BPM[Lw]||"NoteForm.initialValue: null"||{null}
FINE - 2025-05-23T07:37:45.329988||BPM[Lw]||"validating..."||{null}
FINEST - 2025-05-23T07:37:45.330010||BPM[Lw]||"time: true"||{null}
FINEST - 2025-05-23T07:37:45.330019||BPM[Lw]||"note: true"||{null}
FINEST - 2025-05-23T07:37:45.330025||BPM[Lw]||"bp: true"||{null}
FINEST - 2025-05-23T07:37:45.330035||BPM[Lw]||"weight: null"||{null}
FINEST - 2025-05-23T07:37:45.330042||BPM[Lw]||"intake: null"||{null}
FINE - 2025-05-23T07:37:45.330054||BPM[Lw]||"Calling save"||{null}
FINE - 2025-05-23T07:37:45.330083||BPM[Lw]||"validating..."||{null}
FINEST - 2025-05-23T07:37:45.330090||BPM[Lw]||"time: true"||{null}
FINEST - 2025-05-23T07:37:45.330096||BPM[Lw]||"note: true"||{null}
FINEST - 2025-05-23T07:37:45.330102||BPM[Lw]||"bp: true"||{null}
FINEST - 2025-05-23T07:37:45.330107||BPM[Lw]||"weight: null"||{null}
FINEST - 2025-05-23T07:37:45.330110||BPM[Lw]||"intake: null"||{null}
FINER - 2025-05-23T07:37:45.330155||BPM[Lw]||"Saving values: Note(time: 2025-05-23 07:37:13.308298, note: Changed color, color: 4282339765), BloodPressureRecord(time: 2025-05-23 07:37:13.308298, sys: 120, dia: 80, pul: 78), null, null"||{null}
FINER - 2025-05-23T07:37:45.330184||BPM[_Hw]||"Returning result: (Ow: null, Qw: Note(time: 2025-05-23 07:37:13.308298, note: Changed color, color: 4282339765), Sw: BloodPressureRecord(time: 2025-05-23 07:37:13.308298, sys: 120, dia: 80, pul: 78), Uw: 2025-05-23 07:37:13.308298, Vw: null)"||{null}
FINEST - 2025-05-23T07:37:45.330386||BPM[context.EntryUtils]||"received (Ow: null, Qw: Note(time: 2025-05-23 07:37:13.308298, note: Changed color, color: 4282339765), Sw: BloodPressureRecord(time: 2025-05-23 07:37:13.308298, sys: 120, dia: 80, pul: 78), Uw: 2025-05-23 07:37:13.308298, Vw: null) from dialog"||{null}
FINEST - 2025-05-23T07:37:45.380174||BPM[context.EntryUtils]||"mounted=true"||{null}
INFO - 2025-05-23T07:37:45.380219||BloodPressureMonitor||"Instance of 'uz'"||{null}

Mit frdl. Grüßen / Best regards
Jens Mohr
(Gesendet vom Handy /
Sent from mobile phone)

derdilla, 2025-05-27:
Since no one else reported still having issues: I'm happy to hear it suddenly works :). Looks like it might have been a framework / dependency bug then.