derdilla/blood-pressure-monitor-fl #556: Export doesn't export all data

hanskuster, 2025-04-20:
### App version

v1.8.5 and earlier

### Bug Description

I exported the data to csv. My latest data entry in the app is April 20th 2025.
The most recent date in the exported data is January 8th 2025.
In the app I have 2 data entries for every week,  so the export misses about 14 weeks or 28 data entries.
Currently I have 625 data entries in the app and in the exported csv only 593

### Steps to Reproduce

Goto Settings
Export/Import
select "lifetime"
Select export format "csv"
click "EXPORT"
import data into libreOffice

In libreOffice convert all timestampUnixMs with this formula
=DATE(1970;1;1) + (timestampUnixMs/1000)/60/60/24

### Details (Screenshots, Debug Info, Configurations)

<details><summary>Attachments</summary>

</details>

hanskuster, 2025-04-22:
Sorry, I realized this after submitting the issue.
Will try with current version again.
Because of issue with not saving I'm reverted to version 1.8.5

hanskuster, 2025-04-24:
Tested export on version 1.8.6 too.
Had the same result. Still doesn't export all data.

derdilla, 2025-05-22:
You can now (1.8.7) enable very verbose logging (leaks entered values) in Settings -> App Info. Could you try capturing some logs of this bug with some test data and then send them here? No data leaves the app until you press copy.

derdilla, 2025-05-27:
Happy to hear it suddenly works :). Looks like it might have been a framework / dependency bug then.

closed 2025-05-27 by derdilla