derdilla/blood-pressure-monitor-fl #557: Import multiple values

sjaeckel, 2025-04-23:
### What you want:

Would it be possible to import multiple values instead of selecting only one?
Maybe select them somehow and open the import dialog for each of them?

### Why you want it:

First of all thanks a lot for maintaining and improving this project! It's a pleasure to use.

When importing measurements via Bluetooth, and there are multiple values measured since the last import, there's now this dialog to select a value to import.
Currently I don't have a way to import those "missing values", I don't know why, maybe my device sends them only once!? Which means I have to enter them manually ...

Thanks in advance!

derdilla, 2025-04-24:
The spec allows for single and multi-message records to be transmitted. Some devices transmit all measurements (see #352 for details), but there is no reliable way to force them. To add those measurements we would also need to trust the device time which has its own issues (#410).

sjaeckel, 2025-04-24:
Thanks for the feedback.

> The spec allows for single and multi-message records to be transmitted. Some devices transmit all measurements (see [#352](https://github.com/derdilla/blood-pressure-monitor-fl/issues/352) for details), but there is no reliable way to force them.

Interesting! Thanks for the pointers.
I don't believe that we should try to force them or expect any special behavior, but just provide a UI to handle whatever the device sends.

> To add those measurements we would also need to trust the device time which has its own issues ([#410](https://github.com/derdilla/blood-pressure-monitor-fl/issues/410)).

That's also true, but I believe that this isn't a real problem [^1]. First, once #439 is done, we should be able to trust the device time (at least a bit more). And second, that's why I initially proposed the following:

> [...] open the import dialog for each of them.

so the user can adjust the time for each value, if necessary. Maybe a global setting would also make sense "Trust the device data", which then doesn't open the import dialog, but simply adds the values to the list. I propose this, because I've never edited something in the import dialog, but always only clicked on 'Save'.

[^1]: in the background you hear the narrator saying "_that's what he wants to believe, but deep inside he's aware that this **is** a problem_"

derdilla, 2025-04-24:
Ok, you convinced me that this is a nice feature :), although I'm still not entirely sure about the UX details and how to distinct already imported measurements from new ones in case we can't trust the device time.

Now all that's left is to find someone with a bluetooth device and a bunch of time to reliably implement this. (Implementing new bluetooth features is quite low on my personal priority list :P)

sjaeckel, 2025-06-02:
Sorry for the long silence.

> Ok, you convinced me that this is a nice feature :)

🥳 

> although I'm still not entirely sure about the UX details

that's also something I've no clue how to do best ... maybe change the "import button" to a "radio/select button" via a long press?

> how to distinct already imported measurements from new ones in case we can't trust the device time

I'd assume the measurements which are sent by the device always have the same timestamp, so those can already be filtered out. The rest of the decision I'd leave up to the user!?

> Now all that's left is to find someone with a bluetooth device and a bunch of time to reliably implement this. (Implementing new bluetooth features is quite low on my personal priority list :P)

hmm, why is this a bluetooth feature? I only see a UX change!?