derdilla/blood-pressure-monitor-fl #558: Values from Bluetooth not stored

pschichtel, 2025-04-30:
### App version

1.8.6

### Bug Description

When I manually entry records they get stored and the graph starts rendering after at least 2 entries exist.
When I create an entry and use the "Bluetooth input" button it does read the correct values from my device and prefills the form, but when I hit save the app returns to the main screen without saving anything.

I considered it might be just a display bug, so I tried the CSV export, which is also empty. I also tried touching around in the form before saving, but that also didn't make a difference.

I'm one a Nothing Phone (1) with Android 15 (Nothing OS 3.0).

### Steps to Reproduce

1. Install app
2. Connect device (in my case a Medisana BU580 connect)
3. Press the big + button and 
4. Press "Bluetooth input"
5. Press save
6. Nothing saved.

### Details (Screenshots, Debug Info, Configurations)

<details><summary>Attachments</summary>

</details>

pschichtel, 2025-04-30:
it seems my manually inserted entries are also not showing up in the export, so I might also be affected by #556.

pschichtel, 2025-04-30:
This mit also be duplicate of #552, I'll though I'm primarily facing this with bluetooth input.

derdilla, 2025-05-22:
You can now (1.8.7) enable very verbose logging (leaks entered values) in Settings -> App Info. Could you try capturing some logs of this with some test data and then send them here? No data leaves the app until you press copy.

pschichtel, 2025-05-28:
will there be an f-droid release of 1.8.7?

derdilla, 2025-05-29:
Only if they really want it. I messed up the flutter version I maintain for them (#565).

derdilla, 2025-05-30:
Thank you for confirming. So I can't blame my dependencies then :)
Could you, or anyone else, check post the logs as described [above](https://github.com/derdilla/blood-pressure-monitor-fl/issues/558#issuecomment-2902503553) then? (leaks measured value, consider censoring that part)

dymn, 2025-05-30:
Is that enough? Did I do everything right?

[Log.txt](https://github.com/user-attachments/files/20524157/Log.txt)

derdilla, 2025-05-30:
Does your measurement happen to be inserted at 2025-05-25? By the looks of it, #439 has a good chance of fixing this as well

closed 2026-10-29 by derdilla