derdilla/blood-pressure-monitor-fl #565: F-Droid build failed

linsui, 2025-05-23:
https://gitlab.com/fdroid/checkupdates-bot-fdroiddata/-/jobs/10126479261 Could you please take a look? Thanks!

derdilla, 2025-05-23:
F-Droid seems to use Flutter version 3.29.2. The app was upgraded to depend on [3.32.0](https://storage.googleapis.com/flutter_infra_release/releases/stable/linux/flutter_linux_3.32.0-stable.tar.xz).

This is because I forgot to update the version constraint in the pubspec for f-droid. Are you fine with changing this manually for this version in f-droids build file? I would like to avoid polluting the git release tags for this.

linsui, 2025-05-23:
Can you also use the pinned version in the CI? If you just maintain it for F-Droid then it's very likely to forget syncing it with the version you use.

derdilla, 2025-05-23:
Good idea, I can probably do that. I'm also currently working on some shell scripts that should streamline my releases, which should also keep my versions in sync once they are ready.

Since this release was maintenance with a few debug options, there should be no problems with F-droid skipping it.

closed 2025-05-23 by derdilla